riscv-collab/riscv-openocd #6: Compile error: error: left shift of negative value

riktw, 2016-12-01:
Hello,

I got the following error when compiling the openocd tools:

`In file included from /home/rik/Documents/freedom-e-sdk/openocd/src/target/riscv/riscv.c:11:0:
/home/rik/Documents/freedom-e-sdk/openocd/src/target/riscv/riscv.c: In function ‘riscv_examine’:
/home/rik/Documents/freedom-e-sdk/openocd/src/target/riscv/riscv.c:24:69: error: left shift of negative value [-Werror=shift-negative-value]
 #define get_field(reg, mask) (((reg) & (mask)) / ((mask) & ~((mask) << (1))))
                                                                     ^
/home/rik/Documents/freedom-e-sdk/openocd/src/helper/log.h:108:5: note: in definition of macro ‘LOG_DEBUG’
     expr); \
     ^~~~
/home/rik/Documents/freedom-e-sdk/openocd/src/target/riscv/riscv.c:1727:31: note: in expansion of macro ‘get_field’
  LOG_DEBUG("  abussize=0x%x", get_field(dminfo, DMINFO_ABUSSIZE));
`

I had a look and found out it is being caused by the following line:
LOG_DEBUG(" abussize=0x%x", get_field(dminfo, DMINFO_ABUSSIZE));
More specific:
#define DMINFO_ABUSSIZE (0x7f<<25)
If I comment the LOG_DEBUG line it compiles fine.

timsifive, 2016-12-01:
Thanks for digging a little. Does this patch fix it:
```
diff --git a/src/target/riscv/riscv.c b/src/target/riscv/riscv.c
index d4af3de..4ce732b 100644
--- a/src/target/riscv/riscv.c
+++ b/src/target/riscv/riscv.c
@@ -132,7 +132,7 @@ typedef enum slot {
 #define DMCONTROL_FULLRESET            1
 
 #define DMINFO                                 0x11
-#define DMINFO_ABUSSIZE                        (0x7f<<25)
+#define DMINFO_ABUSSIZE                        (0x7fU<<25)
 #define DMINFO_SERIALCOUNT             (0xf<<21)
 #define DMINFO_ACCESS128               (1<<20)
 #define DMINFO_ACCESS64                        (1<<19)
```

timsifive, 2016-12-01:
This is also https://github.com/sifive/freedom-e-sdk/issues/10

timsifive, 2016-12-02:
Please close this issue if it works for you now.

riktw, 2016-12-03:
Thanks, that indeed fixed the issue :)

closed 2016-12-03 by riktw