riscv-collab/riscv-openocd #8: Can't display/i $pc in Scratchpad

mwachs5, 2016-12-14:
I can't currently display instructions which are in Scratchpad memory:

`
(gdb) load
Loading section .text, size 0x848 lma 0x80000000
...
Start address 0x80000000, load size 2194
(gdb)display/i $pc
  0x80000000: <error: Cannot access memory at address 0x80000000>
(gdb) x 0x80000000
0x80000000 <_prog_start>: 0xf14024f3
`

Any thoughts? Is this related to the working area declared at 0x80000000?

If I have a program which is executed from Flash (0x20000000) then it works as expected.

timsifive, 2016-12-19:
This error comes from gdb, somehow. It doesn't ask OpenOCD for anything. Specifying the address works:
```
(gdb) display/i 0x80000000
6: x/i 0x80000000
=> 0x80000000 <_start>:	j	0x8000000c <handle_reset>
```

I fear gdb is trying to access the ELF file, and then failing somehow. I'll dig more once the latest riscv gdb builds again.

mwachs5, 2016-12-19:
Is if because of the working area? If I leave off that it will work. Does
gdb think that area is off limits?

timsifive, 2017-01-02:
What's happening is that gdb gets the PC value (0x8000_0000). Then because the PC register has a type of int32, it gets sign extended into an internal address type (which is 64 bits), to 0xffff_ffff_8000_0000. Then gdb tries to find the memory region corresponding to that address, comes up empty, and prints out the error you see.

The simple fix is to change the type of all GPRs to be unsigned. I don't know if that's good/bad. You'd usually think of the pc as unsigned, because it contains an address. But what about other GPRs? Some contain addresses, and some just values. If a register contains 0xffffffff it would be nice if gdb prints out its value as -1, which it won't if we tell it the value is unsigned.

For comparison, MIPS declares its registers to be signed. I don't know if it has the same problem we have.

My inclination is to make GPRs unsigned, and put a comment in there explaining why it is that way. Then when somebody discovers it's an issue, they can read the rationale and make an informed decisions.

closed 2017-03-02 by palmer-dabbelt