riscv-collab/riscv-openocd #10: Rare assertion error when trying to flash memory

richardxia, 2016-12-23:
I just wanted to report a C assertion error that I saw once when trying to use OpenOCD to load a program into a memory-mapped SPI flash. The relevant commands that I ran on OpenOCD were:

```
flash protect 0 0 last off
program PATH_TO_PROGRAM verify
flash erase_sector 0 64 64
resume 0x20000000
```

The relevant parts of the log were:

```
cleared protection for sectors 0 through 255 on flash bank 0
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (<unknown>), part: 0x0e31, ver: 0x1)
riscv.cpu: target state: halted
halted at 0x404 due to software breakpoint
** Programming Started **
auto erase enabled
riscv.cpu: target state: halted
halted at 0x80000004 due to software breakpoint
riscv.cpu: target state: halted
halted at 0x80000004 due to software breakpoint
riscv.cpu: target state: halted
halted at 0x80000004 due to software breakpoint
riscv.cpu: target state: halted
halted at 0x80000004 due to software breakpoint
riscv.cpu: target state: halted
halted at 0x80000004 due to software breakpoint
riscv.cpu: target state: halted
halted at 0x80000004 due to software breakpoint
Error: ftdi device did not accept all data: 3584, tried 16382
Error: unable to purge ftdi rx buffers: LIBUSB_ERROR_NO_DEVICE
openocd: /openocd/src/jtag/drivers/mpsse.c:849: mpsse_flush: Assertion `ctx->write_count == 0 && ctx->read_count == 0' failed.
```

aswaterman, 2016-12-23:
FYI, this is different from the failure mode I saw with one board (it would hang randomly after a few iterations of halted/breakpoint).

mwachs5, 2017-01-26:
The first error looks like the board went to sleep while it was getting flashed.

timsifive, 2017-01-26:
LIBUSB_ERROR_NO_DEVICE sounds more like a hardware issue than an openocd issue, but I don't have any real insight unfortunately.

mwachs5, 2017-01-26:
we can try to duplicate this by just unplugging the board while we were
trying to flash it.

timsifive, 2017-01-26:
FWIW, in a quick test unplugging the board with openocd just connected just
resulted in openocd hanging. This is from inside a VM, though. Maybe if
you're not in a VM the behavior is different.

Tim

timsifive, 2017-09-18:
When running tests locally, I have noticed that the USB-TINY-H will sometimes fall off the USB, and I have to unplug it and plug it back in. I have never encountered a message mentioning LIBUSB_ERROR_NO_DEVICE.

I'm closing this because there's no action to take to follow up on it. Hopefully if people run into this in the future they're able to supply more debug info, or another indication of what the problem might be.

closed 2017-09-18 by timsifive