riscv-collab/riscv-openocd #14: Unpredictable behavior when reading nonexistent CSRs

mwachs5, 2017-02-05:
OpenOCD doesn't seem to do the right thing when reading CSRs that don't exist.

For example, the HiFive1 board has no CSR 0xF10 (which is where $misa was in Priv 1.9) but it does have $misa at  0x301 (1.9.1 location)

If I launch OpenOCD and connect with GDB:
```
p/x $csr769
p/x $csr3856
```

The first CSR is read successfully with the right $misa value. OpenOCD tries to read the other CSR and gets an exception. This seems like the expected behavior,  but this doesn't translate to letting GDB know that the register doesn't exist, and GDB thinks the register has value 0xFFFFFFFF. This is problematic for something like $misa, because then GDB infers the wrong thing about the core rather than realizing $misa doesn't exist.

I tried looking at the code but not sure where the issue would be.

mwachs5, 2017-02-05:
I think this is actually a bug in openocd. None of the code is actually checking the return value of register_get or register_set.

I'm trying out a fix but no real internet access to push it.

timsifive, 2017-02-05:
ISTR looking into this issue (but not for this reason) and concluded that
OpenOCD does not tell gdb that the access failed.

We could change the gdb code that detects whether C is supported to treat
0xffffffff as an invalid misa. That seems like a very unlikely value. Maybe
on a 128-bit core where we somehow only read the lower 32 bits, and every
extension is implemented, you might see it. That's pretty improbably.

Tim

mwachs5, 2017-02-05:
Right. I got openocd to properly tell GDB it failed (will make a PR later
today) but not sure how that information would actually be used in the GDB
code for deciding the "c" extension. The comments say "try to read misa"
but it doesn't show any way of checking whether the "try" failed.

So I think both fixes are necessary.

aswaterman, 2017-02-06:
Note that MISA is defined to return 0 if not implemented, so maybe using 0
as the error value is also sensible?

mwachs5, 2017-02-06:
Ah, ok. So in general misa CSR doesn't give an exception (would return 0) if not implemented, and this is an unusual case because the register doesn't even exist.

So is misa a special case, or are all CSRs supposed to return 0 vs give an exception when accessed?

aswaterman, 2017-02-06:
misa/mvendorid/marchid are special cases; in general, unimplemented CSRs will cause illegal instruction exceptions.

timsifive, 2017-02-06:
I don't think that helps us in this case. The issue is that gdb asks to
read csrN, which is not where misa lives on this hardware. OpenOCD has no
way of knowing that really what gdb is asking for is misa. (Not without
changing around how the CSRs are handled overall, which is not a good idea.)

The RISC-V OpenOCD code already doesn't return any value. It returns error.
Somehow this ends up in gdb as 0xffffffff. The ideal solution is probably
to tell gdb about different versions of the privileged spec, but that's a
big change too, and (hopefully) unlikely to be useful in the long term.

mwachs5, 2017-02-06:
Right. We could make a one-off solution that hacks around the fact that MISA moved, but really this issue only comes up with talking to cores that are implementing priv 1.9 instead of 1.9.1

timsifive, 2017-02-06:
I'm adding a config option to gdb that allows the user to control whether
compressed breakpoints are used. (Defaults to 'auto'.)

mwachs5, 2017-02-17:
Fixed via #15 , but now this causes problems because GDB attempts to save all registers when calling functions, and if there is no F extension (for example) now GDB gets an error which it doesn't handle. Should GDB read MISA first?

mwachs5, 2017-02-17:
@timsifive is there no other communication from OpenOCD about whether registers exist or not? What is the "exists" field in the register structure for?

timsifive, 2017-02-17:
Possibly. How gdb handles registers is complicated and not documented anywhere that I have found. There is a way where the debug server (OpenOCD) sends XML-formatted data to gdb listing what all registers exist. I think gdb's riscv-tdep.c ignores that list in favor of the one it has baked in. I had spent some time trying to figure out how to use it but didn't get anywhere, and didn't see any obvious advantage at the time.

mwachs5, 2017-02-17:
Ok I opened https://github.com/riscv/riscv-binutils-gdb/issues/60

riscv-tdep seems to have vanished though, not sure where the relevant code is.

timsifive, 2017-07-21:
https://github.com/riscv/riscv-binutils-gdb/issues/60 issue has been fixed. Does that mean we can close this issue as well?

mwachs5, 2017-07-21:
Well, this issue was originally about not printing junk in GDB when you read a register that doesn't exist. It looks like the fix for riscv/riscv-binutils-gdb#60 was to just not read registers that don't exist? If we're fine with OpenOCD's errors being ignored, then I guess it's sufficient.

timsifive, 2017-07-21:
OpenOCD's errors are no longer ignored. I don't remember where that crept in. The GDB fix was actually necessary to make user function calls work at all. (A long time ago they worked and had the side effect of errors being printed out.)

mwachs5, 2017-07-21:
Ah, OK. Then fine to close this.

I wonder how the change in OpenOCD which made this work will play with upstreaming, since it was changing non-RISC-V code

timsifive, 2017-07-21:
There are several non-RISC-V changes and I haven't started the process of trying to push those yet. It will be fun for another day. :-)

closed 2017-07-21 by timsifive