riscv-collab/riscv-openocd #17: Openocd build problem on aarch64

laanwj, 2017-02-24:
Openocd cannot currently be built on aarch64. The platform is not recognized, resulting in an error in the `jimtcl` autosetup.

It was very simple to fix this: replace `configure.guess`  and `configure.sub` in `openocd/jimtcl/autosetup` with recent versions from GNU. E .g.
```bash
cd jimtcl/autosetup
wget -O config.guess 'http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.guess;hb=HEAD'
wget -O config.sub 'http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.sub;hb=HEAD'
```
I noticed this while trying to build the freedom-e-sdk  (see https://github.com/sifive/freedom-e-sdk/issues/51), which has this as a submodule. This goes two projects deeper to [jimtcl](https://github.com/msteveb/jimtcl) then [autosetup](https://github.com/msteveb/autosetup), it may well be solved there and that this is just a matter of updating the submodule.

mwachs5, 2017-02-28:
Thanks for filing this. We will soon be doing a a major rebase onto the latest master OpenOCD, and will make sure this gets handled with that.

timsifive, 2017-07-08:
This rebase has happened. @laanwj can you confirm that it builds now?

laanwj, 2017-07-11:
Yes, I just tried building the tools again, and there was no error. Thanks!

closed 2017-07-11 by laanwj