riscv-collab/riscv-openocd #45: V13: How does GDB user know hart is halted or not?

mwachs5, 2017-05-01:
This might just be an issue of my understanding how to use GDB + OpenOCD as a user.

What should be the interface when GDB connects to a running hart? Is user able to know whether the hart is halted or running? Or should GDB always halt the hart once it connects?

Right now, if GDB connects with OpenOCD to an unhalted hart, the hart doesn't get halted automatically, and it's hart to tell from the GDB side what's going on as you are reading registers/memory on a running hart which doesn't support it. OpenOCD gives ERROR_HALT_RESUME, but that doesn't get reflected in GDB.

palmer-dabbelt, 2017-05-01:
I think OpenOCD should just figure out how to halt the harts as it can't do anything without them halted.

mwachs5, 2017-05-01:
Raw OpenOCD supports reading registers, memory, etc without halting harts. Not sure how this is reflected in GDB.

timsifive, 2017-12-29:
AFAICT gdb's behavior sucks if it connects to a running target. It appears to assume the target is halted, and doesn't attempt to verify this in any way. The best solution seems to be to halt the target in the OpenOCD config script. I'm not sure if there's anything really to do, unless somebody can point us at an example of how it can work better.

mwachs5, 2018-01-03:
Ok, then maybe we should just close this issue.

closed 2018-01-03 by timsifive