riscv-collab/riscv-openocd #46: Make some messages "quiet"

mwachs5, 2017-05-02:
It would be nice to make some messages "quiet". For example, the 'expected' failure for a 32-bit core when you're trying to figure out if it is a 32-bit core would be nice to suppress, so we aren't seeing alarming 'error'. Also, we could suppress the messages while polling.

timsifive, 2017-09-18:
What do you mean by suppressing the messages while polling? AFAICT there is no output during polling when everything is working (unless you enable debugging, and then you deserve what you get).

mwachs5, 2017-09-18:
in the v11, you didn't print stuff out while polling. v13 (at least at the time I filed this) printed out everything while polling, which would quickly overrun debug logs. 

maybe that behavior has already been changed.

timsifive, 2017-12-27:
Is this still an issue? On HiFive1 I get:
```
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive, Inc.), part: 0x0e31, ver: 0x1)
Info : dtmcontrol_idle=5, dbus_busy_delay=1, interrupt_high_delay=0
Info : dtmcontrol_idle=5, dbus_busy_delay=1, interrupt_high_delay=1
Info : dtmcontrol_idle=5, dbus_busy_delay=1, interrupt_high_delay=2
Info : dtmcontrol_idle=5, dbus_busy_delay=1, interrupt_high_delay=3
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=3
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=4
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=5
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=6
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=7
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=8
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=9
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=10
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=12
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=14
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=16
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=18
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=20
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=23
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=26
Info : dtmcontrol_idle=5, dbus_busy_delay=2, interrupt_high_delay=29
Info : [0] Found 2 triggers
halted at 0x800002e8 due to hardware trigger
Info : Examined RISCV core; XLEN=32, misa=0x1105
Info : Listening on port 45885 for gdb connections
Info : Found flash device 'issi is25lp128' (ID 0x0018609d)
cleared protection for sectors 64 through 255 on flash bank 0
Ready for Remote Connections
Info : tcl server disabled
Info : telnet server disabled
Info : accepting 'gdb' connection on tcp/0
```

And on a 0.13 (spike) I get:
```
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:44271
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
Info : JTAG tap: riscv.cpu tap/device found: 0xdeadbeef (mfg: 0x777 (<unknown>), part: 0xeadb, ver: 0xd)
Warn : JTAG tap: riscv.cpu       UNEXPECTED: 0xdeadbeef (mfg: 0x777 (<unknown>), part: 0xeadb, ver: 0xd)
Error: JTAG tap: riscv.cpu  expected 1 of 1: 0x10e31913 (mfg: 0x489 (SiFive, Inc.), part: 0x0e31, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Info : Disabling abstract command reads from CSRs.
Info : Disabling abstract command writes to CSRs.
Info : [0] Found 4 triggers
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, 4 triggers
Info : Listening on port 37507 for gdb connections
```

mwachs5, 2017-12-28:
Users get confused when they see the Error message for detecting the 32 bit
target. Based on the spike log below it looks like that error is no longer
showing up. So probably fine to close this for now.

(By the way to get rid of the other error messages you could just remove
the “expected ID” part of your cfg file)

closed 2017-12-28 by timsifive