riscv-collab/riscv-openocd #49: Segfault when trying to upload to HiFive

tannewt, 2017-05-11:
Hi all,
I'm trying to use my own toolchain to upload a demo to the HiFive but openocd is segfaulting. 

The output without debugging is:
```
$ work/build/openocd/src/openocd -f bsp/env/freedom-e300-hifive1/openocd.cfg
Open On-Chip Debugger 0.10.0-dev-g563f6acc-dirty (2017-05-10-23:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv-013.cpu tap/device found: 0x10e31913 (mfg: 0x489 (<unknown>), part: 0x0e31, ver: 0x1)
Info : dtmcontrol_idle=5, dbus_busy_delay=1, interrupt_high_delay=0
Info : dtmcontrol_idle=5, dbus_busy_delay=1, interrupt_high_delay=1
Info : Examined RISCV core; XLEN=32, misa=0x40001105
Segmentation fault: 11
```

I dug in and debugged some and found that `riscv_halt_one_hart` calls `riscv_is_halted` which attempts to call `r->is_halted` which is null. Here is the last bit of output with my debugging added.

```
Debug: 364 82 riscv.c:683 riscv_openocd_halt(): halting all harts
Debug: 365 82 riscv.c:841 riscv_halt_one_hart(): halting hart 0
Debug: 366 82 riscv.c:952 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 367 82 riscv.c:963 riscv_set_current_hartid(): registers already initialized, skipping
Debug: 368 82 riscv.c:843 riscv_halt_one_hart(): return ok 123
Debug: 369 82 riscv.c:1050 riscv_is_halted(): is halted
Debug: 370 82 riscv.c:1052 riscv_is_halted(): got info 0x1048e4000 0x0 0x1048e4000
Segmentation fault: 11
```

Thanks! I was excited to get this far. :-)

palmer-dabbelt, 2017-05-11:
https://github.com/riscv/riscv-openocd/commit/bcf2a16b0d0d186fdfb371ff4e8009251626ee22

Sorry about that, I'm still in the middle of refactoring the code so I need to use some of the old interfaces for the v11 stuff and I just forgot about them when fixing a recent bug.  This should be fixed, if you see any other SEGVs feel free to open another bug.

closed 2017-05-11 by palmer-dabbelt

tannewt, 2017-05-11:
Thanks for the quick fix! I'll try it tonight hopefully. :-)