riscv-collab/riscv-openocd #51: If read_memory() is called twice one after another, is there something goes wrong?

Rain0Huang, 2017-05-22:
In 'riscv-011.c', if 'read_memory' is called another 'read_memory', the cache line dram_cache[n] will all hit. Then when you execute 'cache_write', an interrupt will be raised as all cache line hit. SO there is the problem, code in debug-ram is run with a interrupt written without our true-address(dram_cache[4] given after 'cache_write' ) !  And now, the value as address(on dram_cache[4]) is unexpected(depend on the data we read out the last time) .
To avoid this, I force dirty cache line by clear dram_cache[4] as zero before 'cache_write' is called. 
I mean do you think this problem a bug or not? 
@palmer-dabbelt @mwachs5  etc.

mwachs5, 2017-05-22:
Seems like the same issue reported here https://groups.google.com/a/groups.riscv.org/forum/m/#!topic/debug/DIVd7sLE1a4

Will look into a fix.

mwachs5, 2017-05-22:
Seems like this case should also check for 'run': https://github.com/riscv/riscv-openocd/blame/riscv/src/target/riscv/riscv-011.c#L873

Rain0Huang, 2017-05-23:
mwachs5，you are right , 'L873' is the reason cause an interrupt.

mwachs5, 2017-05-25:
fixed by #52, thanks for reporting the issue

closed 2017-05-25 by mwachs5