riscv-collab/riscv-openocd #59: openocd dies immediately when connecting to spike

timsifive, 2017-06-08:
This started happening in change 459b39ec67bbbe2ea0a192344c5236518da79c65

timsifive, 2017-06-08:
Output:
```
...
Debug: 175 8 riscv-013.c:966 examine(): dtmcontrol=0x61
Debug: 176 8 riscv-013.c:967 examine():   dmireset=0
Debug: 177 8 riscv-013.c:968 examine():   idle=0
Debug: 178 8 riscv-013.c:969 examine():   dmistat=0
Debug: 179 8 riscv-013.c:970 examine():   abits=6
Debug: 180 8 riscv-013.c:971 examine():   version=1
Debug: 181 9 riscv-013.c:207 scan(): 40b r 00000000 @10 -> F 00000000 @00
Error: 182 9 riscv-013.c:385 dmi_read(): failed read from 0x10, status=2

Error: 183 9 riscv-013.c:392 dmi_read(): Failed read from 0x10; value=0x7fffcda681b8, status=2

```

mwachs5, 2017-06-08:
I am not going to have time to look at this today. Is it in reset or something? Does increasing the retry count on the read fix it?

timsifive, 2017-06-08:
I'm looking at it because it's blocking me (unless we just want to outright revert).
In the future, please run riscv-tests/debug as part of creating a pull request.

mwachs5, 2017-06-08:
Let's set it up on Travis if it needs to be tested as a condition for merge

timsifive, 2017-06-08:
That would be cool, although what if there's a change that requires both
spike and openocd to change at the same time?

Tim

timsifive, 2017-06-08:
https://github.com/riscv/riscv-isa-sim/pull/111 fixes this.

mwachs5, 2017-06-08:
Yeah, that is why the check is done at riscv-tools level.

timsifive, 2017-06-08:
Well, should be. Right now the debug tests aren't running as part of the
travis check. On my todo list for post-64-bit.

Tim

closed 2017-06-08 by timsifive