riscv-collab/riscv-openocd #71: RISC-V target support doesn't build without --disable-werror

TM1234, 2017-06-27:
See here:

https://github.com/gnu-mcu-eclipse/openocd/issues/17

As mentioned there the RISC-V target support code does not build unless --disable-werror is specified at openocd configuration time even though openocd normally (and currently - i.e. openocd 0.10.0+dev development branch) compiles without having to suppress warnings as errors.

Maybe the RISC-V code could be cleaned up to eliminate these warnings?

Thanks.

palmer-dabbelt, 2017-06-27:
Ya, we should fix it.  @timsifive, do you mind taking a look?

ilg-ul, 2017-06-27:
I suggest you also try a run on macOS, clang is also not very happy with the code.

timsifive, 2017-06-28:
Does anybody have experience setting up travis to do 32- and 64-bit builds? That should go a long way towards preventing these problems in the future.

PR #74 (2017-07-06): Fix 32-bit build errors.
I only compiled the source. Didn't have the tooling installed to link.
Hopefully that's good enough.
Fixes #71.
commit 450307b6 (2017-07-06): Fix 32-bit build errors.

I only compiled the source. Didn't have the tooling installed to link.
Hopefully that's good enough.
Fixes #71.
closed 2017-07-06 by timsifive