riscv-collab/riscv-openocd #76: bad behavior when setting more triggers than exist

timsifive, 2017-07-05:
A user reported the following behavior on a core which supports 2 triggers:
Setting 2 triggers works as expected.
Setting a 3rd fails as expected, but afterwards debugging doesn't work right even when one of the breakpoints is deleted.

At the very least we should have a test for this case in riscv-tests/debug, and then of course we should make it pass.

mwachs5, 2017-07-05:
Is this specific to v11?

timsifive, 2017-07-05:
I don't know. But it's a good case to have a test for. Once the test is written we can fix it where it turns out to be broken.

timsifive, 2017-08-07:
This is definitely version-independent, and it's messy somewhere in gdb. Some details here: https://sourceware.org/ml/gdb/2017-07/msg00050.html

timsifive, 2018-04-04:
This sounds different, or at least potentially different. I'd like to look at an OpenOCD debug log if you have one.

TM1234, 2018-04-05:
> But if we add before removing, then continue (even though we only ever "really" have 1 set, at most 2 actually set in HW at a time) eventually after 16 steps of this or so we "run out" of triggers. 

Single stepping can implicitly use two breakpoints if there are two control paths that can be taken - e.g. single stepping a conditional branch instruction.
So if you have 2 triggers, use one for an explicit BP and single step a conditional branch instruction then you are implicitly trying to use 3 triggers.
That's why 2 triggers really is insufficient for debugging.
3 is an effective minimum of you want to be able to single step and set at most one other explicit BP.

TM1234, 2018-04-05:
I don't understand Megan.
If you set one bp explicitly and then single step through a conditional branch then this will need three breakpoints/triggers.
One for the explicit BP and two to cater for the conditional branch's two control paths.

mwachs5, 2018-04-05:
I was using the term “single step” loosely, in very specific rom code that
has no branches

timsifive, 2018-11-07:
This problem may be magically fixed once we can use FSF gdb for everything. I'll attempt to reproduce it when that happens.

timsifive, 2020-03-30:
I think this is fixed now. https://github.com/riscv/riscv-tests/pull/272 tests for it.

closed 2020-03-30 by timsifive