riscv-collab/riscv-openocd #98: CRC sometimes fails on slow targets

timsifive, 2017-08-15:
PR #90 and #92 introduced a bug that causes the DebugCompareSections test to (sometimes?) fail when OpenOCD encounters abstractcs.cmderr=1.

One way to encounter this failure is to run `make all` in riscv-tests/debug. Running several spikes in parallel sometimes lets OpenOCD be too fast and create the error condition.

Another, more reliable way, seems to be to apply the following patch to spike:
```
--- a/riscv/debug_module.cc
+++ b/riscv/debug_module.cc
@@ -1,3 +1,4 @@
+#include <unistd.h>
 #include <cassert>
 
 #include "debug_module.h"
@@ -138,7 +139,10 @@ bool debug_module_t::store(reg_t addr, size_t len, const uint8_t* bytes)
     if (dmcontrol.hartsel == id) {
         if (0 == (debug_rom_flags[id] & (1 << DEBUG_ROM_FLAG_GO))){
           if (dmcontrol.hartsel == id) {
-              abstractcs.busy = false;
+            if (random() > RAND_MAX / 4) {
+              usleep(1);
+            }
+            abstractcs.busy = false;
           }
         }
     }
```

Checking the CRC causes OpenOCD to call read_memory() to read the relevant memory, but I've never seen this failure outside of a CRC computation so maybe there's something additional that tickles this problem.

mwachs5, 2017-08-15:
Does CRC computation use the "riscv_run_algorithm" method?

timsifive, 2017-08-15:
No.

timsifive, 2017-08-15:
With the following patch, MemTestBlock fails (even when adding `-d` to openocd):
```
--- a/riscv/debug_module.cc
+++ b/riscv/debug_module.cc
@@ -1,3 +1,4 @@
+#include <unistd.h>
 #include <cassert>
 
 #include "debug_module.h"
@@ -138,7 +139,10 @@ bool debug_module_t::store(reg_t addr, size_t len, const uint8_t* bytes)
     if (dmcontrol.hartsel == id) {
         if (0 == (debug_rom_flags[id] & (1 << DEBUG_ROM_FLAG_GO))){
           if (dmcontrol.hartsel == id) {
-              abstractcs.busy = false;
+            if (random() > RAND_MAX / 10) {
+              usleep(40);
+            }
+            abstractcs.busy = false;
           }
         }
     }
```

timsifive, 2017-08-15:
Since the M[ lines in the log all match and look right, I suspect the bug might actually be in the memory write code.

mwachs5, 2017-08-15:
What do you mean that "it fails"? Does it read/write the wrong data? Or just not able to complete the command?

timsifive, 2017-09-18:
This was simply an instance of memory read not being reliable. It was fixed by #105.

closed 2017-09-18 by timsifive