riscv-collab/riscv-openocd #101: automate passing `set arch riscv:rv32` to gdb?

ilg-ul, 2017-08-16:
I noticed that when starting GDB for a RISC-V debug session I have to pass a command like:

```
set arch riscv:rv32
```

similarly, when starting debug sessions with ARM devices, setting the architecture is not necessary.

it looks like OpenOCD is able to automatically inform GDB what the architecture is, but this feature is not used for RISC-V devices.

is there anything to do to automate passing the architecture for RISC-V devices too?

ilg-ul, 2017-08-24:
ok, at least we know the cause.

timsifive, 2017-12-27:
https://github.com/riscv/riscv-binutils-gdb/issues/54 was fixed.

closed 2017-12-27 by timsifive

ilg-ul, 2017-12-27:
Thank you, Tim.

When do you think you'll have a new official release of GDB and OpenOCD, with all these goodies?

timsifive, 2017-12-27:
This change is in riscv-tools now.