riscv-collab/riscv-openocd #108: riscv013_get_register can't return an error

timsifive, 2017-09-14:
That's a problem if a user requests to read a non-existent register.

palmer-dabbelt, 2017-09-14:
ya, sorry, I was lazy when implementing some of these.  IIRC OpenOCD didn't pass back the correct error messages anyway, but we should refactor all the function pointers in that structure to be able to return error codes.

mwachs5, 2017-09-29:
This is related to #115 . If we revert the change in gdb_server.c, @palmer-dabbelt is correct that these errors would be meaningless anyway. What *should* happen when a user requests to read a non-existent register? Or should this be made a moot point by some other communication between OpenOCD/GDB (e.g. passing an XML file about which CSRs actually exist)?

timsifive, 2017-09-29:
> What should happen when a user requests to read a non-existent register?

The user should get an error.

Note that the XML exchange happens once, so in order to use that OpenOCD would have to attempt to access every CSR to check whether it exists. (Not the end of the world, but will be pretty slow on simulators.) And a platform might have CSRs that only are accessible part of the time.

ilg-ul, 2017-09-29:
why do you insist on considering the CSRs to belong to the same class as the general registers? they are actually system/peripheral registers. most single core architectures exposes them as memory mapped registers; favouring multi-core, risc-v could not do this and mapped them to a separate address space. but they are not general registers.

what other architectures show all system registers together with the general registers? Cortex-M definitely does **not** do this.

timsifive, 2017-09-29:
As I understand the code, CSRs are not part of the general register group. Only the 32 data registers and the PC are. See https://github.com/riscv/riscv-binutils-gdb/blob/riscv-binutils-2.29/gdb/riscv-tdep.c#L756

ilg-ul, 2017-09-29:
attached are some screenshots from the 'General Registers' view in Eclipse, when debugging the HiFive1 board.

As you can see, not only the float registers are there, but all 4096 CSs are also there.

I don't know if the mis-configuration is in GDB or OpenOCD, but, in my opinion the CSRs **should not** be displayed as 'General Registers'.

[image]
[image]
[image]

mwachs5, 2017-09-29:
So perhaps this issue is more relevant if a user tries to access a floating point register when `F` is not supported (keeping in mind that RISC-V allows `F` to be dynamically turned on and off, or disable the extension entirely...

timsifive, 2017-09-29:
How does Eclipse determine what is a "General Register"?

Note that for FPRs there is some logic in gdb to hide them if the F extension isn't present.

ilg-ul, 2017-09-29:
> How does Eclipse determine what is a "General Register"?

It does not, it simply displays what it gets from GDB. In QEMU, where the xml mechanism is functional, exactly the string used in the xml is seen in Eclipse (I know because I prepared such xmls for QEMU).

> Note that for FPRs there is some logic in gdb to hide them if the F extension isn't present.

then this logic probably happens too late, Eclipse asks GDB about the registers very early and then it remebers it for the entire session.

ilg-ul, 2017-09-29:
> So perhaps this issue is more relevant if a user tries to access a floating point register when F is not supported

this is also a good point. 

since Eclipse does not know that these registers do not exist, scrolling the Registers view to the floating point registers triggers accesses to get them and this generates a lot of errors.

closed 2018-01-19 by timsifive