riscv-collab/riscv-openocd #112: OpenOCD should support targets that don't implement a program buffer

timsifive, 2017-09-18:
It would be really helpful to have an FPGA image or something to test against.

TM1234, 2017-10-04:
What exactly are you looking for in case we (Microsemi) can assist?
Do you have one of our boards?
If so which one?

timsifive, 2017-10-05:
Looking for a RISC-V debug target that does not implement the program buffer, but instead implements the system bus access. The SiFive implementation (which I believe you're using some variant of) doesn't qualify.

I do have an M2GL_M2S_EVAL board, and I regression test against it (mainly to get old trigger coverage) fairly regularly.

cdredd, 2017-10-06:
I can't offer a FPGA image, but I do have a Debug Module design for the Pulpino that uses system bus access instead of program buffer. I am trying to test the design at the moment and am will to try code and feedback issues.

We have tried using the OpenOCD and get an "Unable to write ebreak" error. Scanning the code I had assumed that, currently, only Program Buffer systems are supported, but this comment list seems to imply that system bus is supported. Are non-program buffer systems supported?

timsifive, 2017-10-06:
Non-program buffer systems are not currently supported. I want to support them, but need something to test against.

cdredd, 2017-10-08:
Thanks for the prompt reply. I was worried that would be the answer. Just to clarify, I understand from your statement that non-program buffer support is not started yet. Is that correct?

timsifive, 2017-10-09:
It has not been seriously started. In theory register access as well as run/halt/step should already work.

Also note that when I say non-program-buffer, I'm still referring to a debug interface that complies with the latest draft of the spec. Pulpino used to do something completely different for debug, and I'm not planning on supporting that.

cdredd, 2017-10-11:
We could not get as far as trying "register access and run/halt/step". As part of the initialisation, OpenOCD seems to check that their is enough (any?) program buffer. When it checks the **abstractcs** and finds the **progsize** is zero we get "Unable to write ebreak". This happens early on in the initialisation, in the examine stage I think, but I would need to check.

One thing that occurred to me. Is there any inference  made about program buffer presence from the  IDCODE. For in house testing we are using the SiFive code ATM.

I understand about the Pulpino debugger. The JTAG and their system bus access module has been replaced. The Pulpino Core has a dedicated "Debug Port" that gives run/halt/single step and GP/CS register access. I have also put in translation logic to convert the 0.13 standard for (abstract) register access and halt/run/stepping control to Pulpino Core Debug Port accesses.

timsifive, 2017-10-11:
You're right that OpenOCD currently requires a program buffer for the initial target discovery. As part of some other changes I hope to be fixing that over the next few weeks.

There's no need to rely on IDCODE, since the debug spec already specifies the size/presence of a Program Buffer in the dmstatus register.

cdredd, 2017-10-13:
The fix sounds great. Let me know here and I can try it out. This is all sounding more positive than I thought at the start :).  The program buffer seemed to be heavily relied upon in the start of the code, I thought that removing the dependency would be hard to patch for someone who does not know the code. Adding the system bus access, I guess, might be something I can try to patch.

I'm trying to keep with the 0.13 methodology, but am under pressure to revert to Pulpino design.

Thanks for the clarification on IDCODE.

timsifive, 2017-10-13:
I just pushed my work-in-progress to https://github.com/riscv/riscv-openocd/tree/small_progbuf
It should pass the initial detect and be able to access registers without a program buffer. Memory access does not work at all in this branch, but depending on how eager you are to try something, you might be interested. It should provide a starting point if you want to implement memory access using system bus access.

cdredd, 2017-10-13:
Wow that great Tim. We should be able to give it a try middle of next week. I let you know the results. Thanks for your help.

cdredd, 2017-10-22:
We have given small_progbuf  a try. No issues with the OpenOCD. Just a couple of hardware bugs to fix :(. The main issue was that I was not reporting an error for 64bit register access and this resulted in a fail later in the initialisation performed by OpenOCD. I have done the HW fixes, so we will be retrying next week. I'll report back once I have more news.

timsifive, 2017-10-23:
Great! Thanks for the update. I anticipate merging small_progbuf some time
this week. At that point I think everything will work without a program
buffer except for memory access.

timsifive, 2018-03-01:
This is implemented as of change 3c1c6e059c0129743b3b7f26dcd96e0aa0faa380

closed 2018-03-01 by timsifive