riscv-collab/riscv-openocd #115: Changes to gdb_server.c break debugging of some or all non RISC-V targets

TM1234, 2017-09-22:
The changes to gdb_get_registers_packet(), gdb_set_registers_packet(), gdb_get_register_packet() and gdb_set_register_packet() in this commit:

https://github.com/riscv/riscv-openocd/commit/2ae0078fc723d18822ed50f6f3accd32107acbef#diff-1f3fc53de9d7ac5aa6603362f46a669d

break debugging of some (or maybe all?) non RISC-V targets. The rationale for these changes is also not clear.

For example before these changes (or if these changes are disabled in the latest version of gdb_server.c) debugging of our Cortex-M targets worked fine.
But with these changes enabled Cortex-M (e.g. Cortex-M3 in this case) debugging fails with the following Eclipse error dialog:

```
Error in final launch sequence
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
Remote failure reply: E0E
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
Remote failure reply: E0E
Remote failure reply: E0E
```
and the following OpenOCD log output:
```
Open On-Chip Debugger 0.10.0+dev-00143-g8721a129-dirty (2017-09-22-11:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
M2S090
Info : only one transport option; autoselect 'jtag'
adapter speed: 6000 kHz
cortex_m reset_config sysresetreq
trst_only separate trst_push_pull
do_board_reset_init
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : FlashPro ports available: E200OP5M0
Info : FlashPro port used: E200OP5M0
Info : clock speed 6000 kHz
Info : JTAG tap: M2S090.tap tap/device found: 0x2f8071cf (mfg: 0x0e7 (GateField), part: 0xf807, ver: 0x2)
Info : JTAG tap: M2S090.tap disabled
Info : JTAG tap: M2S090.dap enabled
Info : Cortex-M3 IDCODE = 0x4ba00477
Info : M2S090.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Started by GNU MCU Eclipse
Info : accepting 'gdb' connection on tcp/3333
undefined debug reason 7 - target needs reset
Warn : target not halted

Info : dropped 'gdb' connection
```
In case it helps the verbose (-d) log output is:
```
Open On-Chip Debugger 0.10.0+dev-00143-g8721a129-dirty (2017-09-22-11:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 2 command.c:544 command_print(): debug_level: 3
Debug: 14 2 options.c:217 add_default_dirs(): bindir=D:/msys64/home/tommy.murphy/dev/openocd/openocd-microsemi/openocd/bin
Debug: 15 2 options.c:218 add_default_dirs(): pkgdatadir=D:/msys64/home/tommy.murphy/dev/openocd/openocd-microsemi/openocd/share/openocd
Debug: 16 3 options.c:219 add_default_dirs(): exepath=C:/dev/tools/softconsole/SoftConsole_v5.2.0.8/openocd/bin
Debug: 17 3 options.c:220 add_default_dirs(): bin2data=../share/openocd
Debug: 18 3 configuration.c:42 add_script_search_dir(): adding C:/dev/tools/softconsole/SoftConsole_v5.2.0.8/openocd/bin/../share/openocd/scripts
Debug: 19 4 configuration.c:42 add_script_search_dir(): adding C:\Users\tommy.murphy\AppData\Roaming/OpenOCD
Debug: 20 4 configuration.c:42 add_script_search_dir(): adding C:/dev/tools/softconsole/SoftConsole_v5.2.0.8/openocd/bin/../share/openocd/site
Debug: 21 4 configuration.c:42 add_script_search_dir(): adding C:/dev/tools/softconsole/SoftConsole_v5.2.0.8/openocd/bin/../share/openocd/scripts
Debug: 22 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_gdb_port 3333
Debug: 23 8 command.c:143 script_debug(): command - gdb_port ocd_gdb_port 3333
Debug: 25 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_telnet_port 4444
Debug: 26 9 command.c:143 script_debug(): command - telnet_port ocd_telnet_port 4444
[... 381 lines cut ...]
Debug: 376 1048 gdb_server.c:2868 gdb_input_inner(): received packet: '?'
User : 377 1048 gdb_server.c:153 gdb_last_signal(): undefined debug reason 7 - target needs reset
Debug: 378 1048 gdb_server.c:2868 gdb_input_inner(): received packet: 'qXfer:threads:read::0,fff'
Debug: 379 1048 gdb_server.c:2868 gdb_input_inner(): received packet: 'Hc-1'
Debug: 380 1049 gdb_server.c:2868 gdb_input_inner(): received packet: 'qC'
Debug: 381 1049 gdb_server.c:2868 gdb_input_inner(): received packet: 'qAttached'
Debug: 382 1052 gdb_server.c:2868 gdb_input_inner(): received packet: 'g'
Debug: 383 1053 gdb_server.c:1174 gdb_get_registers_packet(): Couldn't get register r0.
Debug: 384 1053 gdb_server.c:1386 gdb_error(): Reporting -304 to GDB as generic error
Debug: 385 1053 gdb_server.c:1022 gdb_connection_closed(): GDB Close, Target: M2S090.cpu, state: reset, gdb_actual_connections=0
Debug: 386 1054 target.c:1561 target_call_event_callbacks(): target event 6 (gdb-end)
Debug: 387 1055 target.c:1561 target_call_event_callbacks(): target event 24 (gdb-detach)
Debug: 388 1056 target.c:4495 target_handle_event(): target: (0) M2S090.cpu (cortex_m) event: 24 (gdb-detach) action: 
	# resume execution on debugger detach
	resume

Debug: 389 1056 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_resume
Debug: 390 1056 command.c:143 script_debug(): command - resume ocd_resume
Debug: 392 1058 target.c:1561 target_call_event_callbacks(): target event 3 (resume-start)
Warn : 393 1058 cortex_m.c:693 cortex_m_resume(): target not halted
Debug: 394 1058 command.c:626 run_command(): Command failed with error code -304
User : 395 1059 command.c:544 command_print(): 

Info : 396 1059 server.c:533 server_loop(): dropped 'gdb' connection
```

timsifive, 2017-09-22:
I don't have time to really dig into this right now, but I can provide a little background.

The change in question causes errors encountered while reading/writing registers to be propagated to gdb. Previously if a register could not be read (eg. because it doesn't exist) gdb would think the read was successful, and we'd have to provide some value that is hopefully recognized as bogus.

In the log you provided, the new code tells you what register couldn't be accessed. It returns an error to gdb, which in turn decides to abort the session. (I don't know why gdb decides to do so.) I've highlighted that part below:
```
Debug: 382 1052 gdb_server.c:2868 gdb_input_inner(): received packet: 'g'
Debug: 383 1053 gdb_server.c:1174 gdb_get_registers_packet(): Couldn't get register r0.
Debug: 384 1053 gdb_server.c:1386 gdb_error(): Reporting -304 to GDB as generic error
Debug: 385 1053 gdb_server.c:1022 gdb_connection_closed(): GDB Close, Target: M2S090.cpu, state: reset, gdb_actual_connections=0
```

The 'g' packet is gdb requesting to see "all" registers, so it's not even a case of gdb requesting a register that doesn't exist. Maybe the target is running and that is why the request fails. It would need more debugging to figure out.

mwachs5, 2017-09-22:
@TM1234 Can you provide the log file on your non RISC-V target without this change applied? Do you still get the error message that @timsifive identified, but it is silently ignored?

At any rate, I suspected that this might cause issues with upstream OpenOCD, since it is returning errors in places that it used to ignore them. I think we should file an issue upstream, since it seems like bad behavior to drop error messages.

TM1234, 2017-09-23:
Hi Megan - attached is the openocd -d log for a debug session with a Cortex-M3 target with the aforementioned gdb_server.c mods disabled. I don't see any error being ignored - do you?

Tim - I wonder if the "Couldn't get register r0" happens because the target is not in debug halt at that stage but the mods to gdb_server.c don't take account of this being a possible reason for not being able to read the regs?

Hope this helps.

[m3-debug.log](https://github.com/riscv/riscv-openocd/files/1326520/m3-debug.log)

TM1234, 2017-09-23:
Sorry - I missed that Tim had posted this earlier:

> The 'g' packet is gdb requesting to see "all" registers, so it's not even a case of gdb requesting a register that doesn't exist. Maybe the target is running and that is why the request fails. It would need more debugging to figure out.

In this specific case I think that this is the problem alright but the point is that the mods mentioned are assuming that the regs can be read and/or that the target is halted when this may not be the case - hence the mods are not really safe.

They work for RISC-V because in many cases the openocd debug config script will have a halt but not all target scripts will have this - nor should they. In fact we have a halt in our RISC-V debug script only because our RISC-V target does not provide the debugger with the ability to do a proper reset/halt via JTAG.

mwachs5, 2017-09-29:
It sounds like we should revert this change to gdb_server.c, and understand why it was not fixed in upstream openOCD. This issue that this change was initially set up to handle has since been handled by a workaround in GDB (you can now explicitly set whether your core supports 'C' extension, even if MISA is not in the 1.9.1 location).

timsifive, 2017-09-29:
This is a more general issue, though. When a user asks to read a CSR, there really isn't a good alternative besides having the debugger try to do the read, and return either success or error. Without this change, the best we can do is return some value like 0 or ~0 and hope the user realizes the CSR isn't present.

mwachs5, 2017-09-29:
I agree that this is a problem. I don't understand why it works for non RISC-V Targets.

timsifive, 2017-09-29:
Does it work for non-RISC-V targets? Eg. MIPS has a coprocessor register space much like CSRs. How is the problem solved there? A user should be able to request to view an arbitrary one, and get back either its value or some message indicating it doesn't exist. And that message should be true (ie. allowing for cores which implement unknown-to-the-debugger extensions).

TM1234, 2017-09-29:
> I agree that this is a problem. I don't understand why it works for non RISC-V Targets

Hi Megan - can you clarify what you mean please?
Your don't understand why what works for non RISC-V targets?

Seems to me that the mods to gdb_server.c are assuming that the registers are always readable.
I'm not sure that this is the case for all targets all of the time - e.g. when not in debug halt?
Seems to be the case for Cortex-M that the regs cannot be read when the target is not in debug halt -  but I can't remember the details of how CoreSight works in this respect offhand.
If the code is assuming that the registers are always readable and/or that the target is in debug halt then  these are not safe assumptions as far as I can tell (and the fact that Cortex-M debugging fails when it works without these mods is a specific case in point)..

Hope this helps.

mwachs5, 2017-09-29:
The mods to gdb_server.c simply return the error that OpenOCD has taken the time to report. If it is not an error for the registers to be read when a given target is not halted, then OpenOCD wouldn't return an error, and the gdb_server.c mods would have no effect. If it IS an error for the registers to be read when the hart is halted, well, then the upstream tool/script/GDB should be notified and handle it properly. It sounds like the real issue is the later : for other targets, the upstream tools or scripts (I don't know which) are doing bad things and never realized it before, so they haven't properly handled the error.

This particular gdb_server code makes no assumptions about whether the registers can be read or not. If an error is returned, it simply passes it on.

TM1234, 2017-09-29:
OK - I don't really understand why these changes break Cortex-M debugging but it works without them.
I don't know if the two logs posted earlier shed any light on this?

mwachs5, 2017-09-29:
I looked at the two logs. In the "passing" log, the 'g' packet which fails on the "failing" log seems to be completely ignored. GDB closes down the session when the error from the `g` packet is passed back in the failing log. So the question is really why does the ARM GDB close down as soon as it gets an error. perhaps there is a "gentler" error type that can be returned.

ilg-ul, 2017-10-04:
for the records:

https://github.com/gnu-mcu-eclipse/openocd/commit/da43038f866cd7444405edd01780fd3316ad187c

mwachs5, 2017-10-04:
Is there a way to ask about this to upstream OpenOCD? Not sure their issue/ Q&A mechanism.

TM1234, 2017-10-04:
> Is there a way to ask about this to the OpenOCD maintainers? Not sure their issue/ Q&A mechanism.

Any use?

http://openocd.org/discussion/

ilg-ul, 2017-10-04:
> why does the ARM GDB close down as soon as it gets an error. perhaps there is a "gentler" error type that can be returned to GDB so it would just ignore the error on its side.

is this the question you want to ask? isn't this more of a GDB question?

on the other side, if the discussion is only about the 4 patches that I just disabled, I think that we are 'fighting against the wind mills', the chances of pushing something like this upstream are not that high.

ilg-ul, 2018-01-19:
Tim, do you have a plan how to fix this?

Did you identify the reasons why these patches were applied?

Personally I would reverse the patches completely, since both myself and Microsemi use OpenOCD like this without any problems.

If you have serious reasons to maintain these patches, in order to make them harmless for other platforms, and make them upstreamable, I suggest you add a new flag to the target structure (like is_error_strict, or whatever), and do the extra processing only when this flag is set, otherwise revert to the original behaviour.

TM1234, 2018-01-19:
I fully agree with Liviu - I had been meaning to revisit this issue but this has nudged me into action. As stated simply earlier several times the changes break at least Cortex-M (and maybe other CPU) debugging but RISC-V debugging (so far) works fine with the changes disabled so they seem to me to be redundant. And they will definitely not be upstreamable as they stand. I vote for their removal too. But if they have to be retained (and the case for retention is very weak as far as I can see) then they should be made optional/configurable as Liviu suggests.

timsifive, 2018-01-19:
On Fri, Jan 19, 2018 at 1:09 AM, Liviu Ionescu <notifications@github.com>
wrote:

> If you have serious reasons to maintain these patches, in order to make
> them harmless for other platforms, and make them upstreamable, I suggest
> you add a new flag to the target structure (like [...]
That's my plan.

closed 2018-03-02 by timsifive

ilg-ul, 2018-03-02:
thank you, Tim. I'll include it in my next release. it would be great to upstream the patch too.

TM1234, 2018-03-26:
This is basically what I said a good while back:

> Matthias Welwarsky
> Mar 24 8:26 AM
> ↩
> Patch Set 1: Code-Review-1
> 
> Ok, but I don't see why a change in generic code is required because the RISC-V support (which is not even in gerrit yet) is broken.

I have never understood why ANY changes were required to generic/non target specific code to accommodate RISC-V changes - and unfortunately commit comments never really explained either.

ilg-ul, 2018-04-10:
The modified patch was cherry picked on Apr 7.

http://openocd.zylin.com/gitweb?p=openocd.git;a=commitdiff;h=2e2bb14b276f5bd973308dcfabd1b8018e187243

Tim, can you confirm that after the latest merge from master, everything is ok?

timsifive, 2018-04-10:
I don't have an ARM target so I can't reproduce this problem. As far as I'm aware both mainline and the RISC-V branch now have the `gdb_report_register_access_error` command.

ilg-ul, 2018-04-10:
The question was "after the latest merge from master, everything is ok for RISC-V targets?"

Do you automatically initialise the flag related to `gdb_report_register_access_error` for RISC-V targets?

timsifive, 2018-04-10:
Just like `gdb_report_data_abort`, it's left up to the config file to enable `gdb_report_register_access_error`. I enable it in my test scripts and it works well.

ilg-ul, 2018-04-10:
So, by default, both are disabled, for all targets.

timsifive, 2018-04-10:
That's right. Both options default to disabled.

PR #219 (2018-03-02): Only propagate register errors on some targets
Without this change, connecting to ARM targets is impossible.

Fixes #115.

Change-Id: Ie33c7e15ac1bed8c9cbd8e6a78de92d5498c5999
commit 1d9418fb (2018-03-02): Only propagate register errors on some targets

Without this change, connecting to ARM targets is impossible.

Fixes #115.

Change-Id: Ie33c7e15ac1bed8c9cbd8e6a78de92d5498c5999