riscv-collab/riscv-openocd #117: openocd locks up if it cannot connect to target

TM1234, 2017-09-28:
I was trying something with what turned out to be a faulty RISC-V target core - i.e. the debug block was not working for some reason. Because of this openocd could not get a debug connection which is fair enough. However what is not good is that in attempting to connect it locked up and could not be terminated other than by pulling the JTAG probe USB cable. Shouldn't the RISC-V openocd code time out at some stage when it cannot connect rather than locking up?

```
Tommy.Murphy@DUBLTC0016 C:\dev\tools\softconsole\SoftConsole_v5.2.0.9\openocd\bin
> openocd -f board/microsemi-riscv.cfg
2017-09-28 10:26:43.604 : Open On-Chip Debugger 0.10.0+dev-00143-g8721a129-dirty (2017-09-27-10:41)
Licensed under GNU GPL v2
2017-09-28 10:26:43.604 : For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
2017-09-28 10:26:43.604 : Info : only one transport option; autoselect 'jtag'
2017-09-28 10:26:43.620 : adapter speed: 6000 kHz
2017-09-28 10:26:43.620 : microsemi_flashpro tunnel_jtag_via_ujtag off
2017-09-28 10:26:43.620 : trst_only separate trst_push_pull
2017-09-28 10:26:43.620 : do_board_reset_init2017-09-28 10:26:43.620 :
2017-09-28 10:26:43.635 : Info : Listening on port 6666 for tcl connections
2017-09-28 10:26:43.635 : Info : Listening on port 4444 for telnet connections
2017-09-28 10:26:44.031 : Info : FlashPro ports available: S2011K2D6Q
2017-09-28 10:26:44.031 : Info : FlashPro port used: S2011K2D6Q
2017-09-28 10:26:44.594 : Info : clock speed 6000 kHz
2017-09-28 10:26:44.594 : Info : JTAG tap: FPGA.tap tap/device found: 0x2f8071cf (mfg: 0x0e7 (GateField), part: 0xf807, ver: 0x2)
2017-09-28 10:26:44.594 : microsemi_flashpro tunnel_jtag_via_ujtag on
2017-09-28 10:26:44.594 : Info : JTAG tap: FPGA.tap disabled
2017-09-28 10:26:44.609 : Info : JTAG tap: FPGA.dap enabled
2017-09-28 10:26:44.609 : Info : RISC-V IDCODE = 0x10e31913
2017-09-28 10:26:44.609 : Info : dtmcontrol_idle=5, dmi_busy_delay=1, ac_busy_delay=0
2017-09-28 10:26:44.609 : Info : dtmcontrol_idle=5, dmi_busy_delay=2, ac_busy_delay=0
2017-09-28 10:26:44.624 : Info : dtmcontrol_idle=5, dmi_busy_delay=3, ac_busy_delay=0
2017-09-28 10:26:44.624 : Info : dtmcontrol_idle=5, dmi_busy_delay=4, ac_busy_delay=0
2017-09-28 10:26:44.624 : Info : dtmcontrol_idle=5, dmi_busy_delay=5, ac_busy_delay=0
2017-09-28 10:26:44.640 : Info : dtmcontrol_idle=5, dmi_busy_delay=6, ac_busy_delay=0
2017-09-28 10:26:44.640 : Info : dtmcontrol_idle=5, dmi_busy_delay=7, ac_busy_delay=0
2017-09-28 10:26:44.640 : Info : dtmcontrol_idle=5, dmi_busy_delay=8, ac_busy_delay=0
2017-09-28 10:26:44.656 : Info : dtmcontrol_idle=5, dmi_busy_delay=9, ac_busy_delay=0
2017-09-28 10:26:44.656 : Info : dtmcontrol_idle=5, dmi_busy_delay=10, ac_busy_delay=0
2017-09-28 10:26:44.671 : Info : dtmcontrol_idle=5, dmi_busy_delay=12, ac_busy_delay=0
2017-09-28 10:26:44.671 : Info : dtmcontrol_idle=5, dmi_busy_delay=14, ac_busy_delay=0
2017-09-28 10:26:44.671 : Info : dtmcontrol_idle=5, dmi_busy_delay=16, ac_busy_delay=0
2017-09-28 10:26:44.687 : Info : dtmcontrol_idle=5, dmi_busy_delay=18, ac_busy_delay=0
2017-09-28 10:26:44.687 : Info : dtmcontrol_idle=5, dmi_busy_delay=20, ac_busy_delay=0
2017-09-28 10:26:44.687 : Info : dtmcontrol_idle=5, dmi_busy_delay=23, ac_busy_delay=0
2017-09-28 10:26:44.702 : Info : dtmcontrol_idle=5, dmi_busy_delay=26, ac_busy_delay=0
2017-09-28 10:26:44.702 : Info : dtmcontrol_idle=5, dmi_busy_delay=29, ac_busy_delay=0
2017-09-28 10:26:44.718 : Info : dtmcontrol_idle=5, dmi_busy_delay=32, ac_busy_delay=0
2017-09-28 10:26:44.718 : Info : dtmcontrol_idle=5, dmi_busy_delay=36, ac_busy_delay=0
2017-09-28 10:26:44.718 : Info : dtmcontrol_idle=5, dmi_busy_delay=40, ac_busy_delay=0
...
and so on until the JTAG probe USB cable was pulled.
```

TM1234, 2017-09-28:
Is the issue in static uint64_t dmi_read(struct target *target, uint16_t address)?

https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c

There seem to be many other cases of 256 iteration loops that may not have any timeout/quit capability meaning that they too could lock up openocd if something goes wrong?

timsifive, 2017-12-28:
I cannot tell from this log where exactly the problem was. dmi_read() does not loop forever, as you point out. If there are infinite loops left we should fix them.
I assume you can't easily reproduce this with more logging.

TM1234, 2017-12-28:
Let me try with an up to date build of openocd.

closed 2018-03-03 by timsifive

TM1234, 2018-03-03:
What was the resolution/fix here?
I don't see any commit referenced.

timsifive, 2018-03-05:
I should've clarified:
Closed because I can't reproduce it.

Please feel free to reopen this with instruction on how to reproduce the problem.