riscv-collab/riscv-openocd #124: --disable-werror should not be required in order for openocd to compile successfully

TM1234, 2017-10-04:
As a general rule the code should build without warnings being ignored as happens when configure --disable-werror is used. The master openocd source base builds without --disable-werrror. At the moment and for a long time now the RISC-V openocd branch has required --disable-werror to build. I will add specific examples of the code that breaks the build when --disable-werror is not used below. Note that my reference build platform here is MinGW-64/MSYS2 32 bit but I believe that the issue affects any other build platform and --disable-werror is required to get the code building successfully.

ilg-ul, 2017-10-04:
clang builds on macOS were also problematic with the RISC-V code when I merged it into my fork.

timsifive, 2017-10-04:
I'm happy to fix any warning that you can find. I believe that the riscv-openocd build happens with -Werror. If not, let's fix that.

`riscv-tools` does use --disable-werror, I believe. That is because we don't know what exact compiler a user will have, and we don't want the build to fail just because this compiler warns about something that the one we're using does not.

TM1234, 2017-10-04:
Apologies for the delay - here are details of the issues that I have encountered.

In this case my build platform is Windows MinGW-64/MSYS2 32 bit console using gcc.

Using these commands a checkout of the master openocd development branch (http://openocd.org/getting-openocd/) builds fine with no errors:

```
./bootstrap
./configure --prefix=`pwd`/openocd
make -j4 -O
```

But building the RISC-V openocd fails...

```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -D__USE_MINGW_ANSI_STDIO -I./src -I./src -I./src/helper -DPKGDATADIR=\"/home/Tommy.Murphy/dev/openocd/openocd-sifive-test/openocd/share/openocd\" -DBINDIR=\"/home/Tommy.Murphy/dev/openocd/openocd-sifive-test/openocd/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/flash/nor/stm32f2x.lo -MD -MP -MF src/flash/nor/.deps/stm32f2x.Tpo -c src/flash/nor/stm32f2x.c -o src/flash/nor/stm32f2x.o
src/flash/nor/stm32f2x.c: In function 'stm32x_handle_unlock_command':
src/flash/nor/stm32f2x.c:1324:67: error: left shift of negative value [-Werror=shift-negative-value]
   stm32x_info->option_bytes.optcr2_pcrop = OPTCR2_PCROP_RDP | (~1 << bank->num_sectors);
                                                                   ^~
```

The above problem is fixed in the master openocd source base:

`		stm32x_info->option_bytes.optcr2_pcrop = OPTCR2_PCROP_RDP | (~1U << bank->num_sectors);
`

And then I get this one:

```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -D__USE_MINGW_ANSI_STDIO -I./src -I./src -I./src/helper -DPKGDATADIR=\"/home/Tommy.Murphy/dev/openocd/openocd-sifive-test/openocd/share/openocd\" -DBINDIR=\"/home/Tommy.Murphy/dev/openocd/openocd-sifive-test/openocd/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/target/riscv/riscv-011.lo -MD -MP -MF src/target/riscv/.deps/riscv-011.Tpo -c src/target/riscv/riscv-011.c -o src/target/riscv/riscv-011.o
In file included from ./config.h:334:0,
                 from src/target/riscv/riscv-011.c:11:
src/target/riscv/riscv-011.c: In function 'read_memory':
./src/helper/replacements.h:30:25: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
 #define MIN(a, b) (((a) < (b)) ? (a) : (b))
                         ^
src/target/riscv/riscv-011.c:2101:29: note: in expansion of macro 'MIN'
   unsigned int batch_size = MIN(count + 3 - i, max_batch_size);
                             ^
./src/helper/replacements.h:30:38: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
 #define MIN(a, b) (((a) < (b)) ? (a) : (b))
                                      ^
src/target/riscv/riscv-011.c:2101:29: note: in expansion of macro 'MIN'
   unsigned int batch_size = MIN(count + 3 - i, max_batch_size);
                             ^
src/target/riscv/riscv-011.c: In function 'write_memory':
./src/helper/replacements.h:30:25: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
 #define MIN(a, b) (((a) < (b)) ? (a) : (b))
                         ^
src/target/riscv/riscv-011.c:2262:29: note: in expansion of macro 'MIN'
   unsigned int batch_size = MIN(count + 2 - i, max_batch_size);
                             ^
./src/helper/replacements.h:30:38: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
 #define MIN(a, b) (((a) < (b)) ? (a) : (b))
                                      ^
src/target/riscv/riscv-011.c:2262:29: note: in expansion of macro 'MIN'
   unsigned int batch_size = MIN(count + 2 - i, max_batch_size);
                             ^
cc1.exe: all warnings being treated as errors
make[2]: *** [Makefile:2896: src/target/riscv/riscv-011.lo] Error 1
```

I addressed this locally as follows in two places in riscv-011.c:

```
#if BUILD_MICROSEMI_MODS == 1
	const uint32_t max_batch_size = 256;
#else
	const int max_batch_size = 256;
#endif
```

I believe that I had other warnings/errors in the past but they must have been fixed in the meantime.

Also bear in mind what Liviu said about further problems (?) when compiling with llvm/clang.

Hope this helps.

PR #126 (2017-10-04): Fix compile warnings.
Partly fixes #124.
commit 24658db5 (2017-10-04): Fix compile warnings.

Partly fixes #124.

Change-Id: I3a7fd65c643e40b142709806cb9fb4cc62bb955f
closed 2017-10-04 by timsifive