riscv-collab/riscv-openocd #128: jtag_vpi change needs to be upstreamed

timsifive, 2017-10-08:
@mwachs5 changed jtag_vpi.c in change 7927e90a. I don't understand the details, so can't upstream this by myself.

mwachs5, 2017-10-08:
> Assigned #128 <https://github.com/riscv/riscv-openocd/issues/128> to

I opened a PR through the OpenOCD system a while ago but seems it failed
some check. I will revisit it.

mwachs5, 2018-05-15:
This is low-priority for me now because we have switched to only using the simpler, existing remote-bitbang interface

ilg-ul, 2018-05-15:
If this patch is no longer needed, better simplify things and revert it.

juliusbaxter, 2018-10-22:
I experienced the issue @mwachs5 was fixing today.

The bug is a stray cycle with TMS high after we've got to RunTestIdle. This causes a couple of different failures, suffice to say it breaks everything.

@mwachs5 do you have a pointer to the upstream failure you saw?

olofk, 2020-08-20:
Is this still an issue? I didn't really understand if we need to patch the verilog/VPI and/or the OpenOCD driver? Is [this](https://github.com/fjullien/jtag_vpi/commit/bbdf4f62642031c03bd984826b9bc7389205e6cb) related?

JanMatCodasip, 2020-08-20:
My understanding is that the stray TMS cycle was this issue: http://openocd.zylin.com/#/c/5151/

It was fixed in the main OpenOCD upstream year ago and that same patch is now part of riscv-openocd, too.

So if I understand this right, this is no longer an issue and can be closed.

nmeum, 2020-09-03:
Hope you don't mind me asking this here but what is the current status regarding upstream integration of riscv support in general? What's prevent riscv support from being integrated upstream?

timsifive, 2020-09-03:
Somebody just needs to do the work. I'm currently working on upstreaming a large chunk at http://openocd.zylin.com/#/c/5821/5. Hopefully I'll have enough time this month to at least complete that part.

closed 2021-07-19 by timsifive