riscv-collab/riscv-openocd #136: monitor command: reg

ilg-ul, 2017-10-31:
Here is a snapshot of the console log of J-Link GDB Server:

```
Received monitor command: regs
X0 = 00000000, X1 = 404000E8, X2 = 80000FE0, X3 = 800007C0
X4 = 00000000, X5 = 00000000, X6 = 00000002, X7 = 00000000
X8 = 0006A6FA, X9 = 00000000, X10= 0006A749, X11= 00000000
X12= 0006A88A, X13= 00000000, X14= 000000FF, X15= 00000000
X16= 0200C000, X17= 00000000, X18= 02000080, X19= 00000000
X20= 00000000, X21= 00000000, X22= 00000000, X23= 00000000
X24= 00000000, X25= 00000000, X26= 00000000, X27= 00000000
X28= 20005000, X29= 0000003D, X30= 000000FF, X31= 00000000
PC = 404004FC
```

On OpenOCD, a similar command returns:

```
===== RISC-V Registers
(0) x0 (/32)
(1) x1 (/32)
(2) x2 (/32)
(3) x3 (/32)
(4) x4 (/32)
(5) x5 (/32)
(6) x6 (/32)
(7) x7 (/32)
(8) x8 (/32)
(9) x9 (/32)
(10) x10 (/32): 0x800000A8
(11) x11 (/32): 0x000002EE
(12) x12 (/32)
(13) x13 (/32)
(14) x14 (/32)
(15) x15 (/32)
(16) x16 (/32)
(17) x17 (/32)
(18) x18 (/32)
(19) x19 (/32)
(20) x20 (/32)
(21) x21 (/32)
(22) x22 (/32)
(23) x23 (/32)
(24) x24 (/32)
(25) x25 (/32)
(26) x26 (/32)
(27) x27 (/32)
(28) x28 (/32)
(29) x29 (/32)
(30) x30 (/32)
(31) x31 (/32)
(32) pc (/32): 0x404004FC
(33) f0 (/32)
(34) f1 (/32)
(35) f2 (/32)
(36) f3 (/32)
(37) f4 (/32)
(38) f5 (/32)
(39) f6 (/32)
(40) f7 (/32)
(41) f8 (/32)
(42) f9 (/32)
(43) f10 (/32)
(44) f11 (/32)
(45) f12 (/32)
(46) f13 (/32)
(47) f14 (/32)
(48) f15 (/32)
(49) f16 (/32)
(50) f17 (/32)
[... 4091 lines cut ...]
(4142) csr4077 (/32)
(4143) csr4078 (/32)
(4144) csr4079 (/32)
(4145) csr4080 (/32)
(4146) csr4081 (/32)
(4147) csr4082 (/32)
(4148) csr4083 (/32)
(4149) csr4084 (/32)
(4150) csr4085 (/32)
(4151) csr4086 (/32)
(4152) csr4087 (/32)
(4153) csr4088 (/32)
(4154) csr4089 (/32)
(4155) csr4090 (/32)
(4156) csr4091 (/32)
(4157) csr4092 (/32)
(4158) csr4093 (/32)
(4159) csr4094 (/32)
(4160) csr4095 (/32)
(4161) priv (/8)
```

Could you do something to avoid displaying this huge list? 

The Eclipse debug plug-ins issue this command at certain important moments, for informative reasons only, to keep a log of processor register values, and the huge response makes the entire console useless, since it usually overflows.

timsifive, 2018-01-02:
As of f13093fff7fca16681e4ded390c82b56c0549bee, unnamed CSRs no longer show up unless the user explicitly configures them to show up.

closed 2018-01-02 by timsifive

ilg-ul, 2018-01-02:
I'll test this on Eclipse as soon as I finish to update all tools to the latest versions and confirm it is ok.