riscv-collab/riscv-openocd #141: autoreconf -i fails

ben-k, 2017-11-28:
My `riscv-tools` installation is failing at this step - I see the same error both at the tip of the `riscv` branch of `riscv-openocd` (7a6704c) and at the version submoduled by riscv-tools (f41652798).

```
$ autoconf --version
autoconf (GNU Autoconf) 2.69
$ automake --version
automake (GNU automake) 1.15
$ autoreconf -i
...
configure.ac:12: error: possibly undefined macro: AC_MSG_WARN
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:36: error: possibly undefined macro: AC_DISABLE_SHARED
configure.ac:37: error: possibly undefined macro: AC_PROG_LIBTOOL
configure.ac:204: error: possibly undefined macro: AC_DEFINE
configure.ac:608: error: possibly undefined macro: AC_MSG_NOTICE
autoreconf: /home/utils/autoconf-2.69/bin/autoconf failed with exit status: 1
```

timsifive, 2017-11-28:
https://stackoverflow.com/questions/8811381/possibly-undefined-macro-ac-msg-error
suggests installing the pkg-config package. Do you have that one installed?

ben-k, 2017-11-29:
Yes.  To confirm a few other packages that my unsuccessful Google-based attempts to solve this turned up:

```
$ pkg-config --version
0.29.2
$ libtoolize --version
libtoolize (GNU libtool) 2.4.6
$ gettext --version
gettext (GNU gettext-runtime) 0.18.2
```

timsifive, 2017-12-01:
I don't know how to solve your problem, but I just want to confirm that my tool versions are identical to yours, except for gettext (I have 0.19.8.1). So that's not likely to be the problem.

closed 2017-12-27 by timsifive

morse-michael, 2018-05-21:
@ben-k: just hit the same issue trying to install these tools on a Centos6 machine with default autoconf (2.63) and autoconf268 installed - in case you have a similar setup please run the following as a WAR:

```
# Autoconf hack - we need to link autoconf268 to autoconf to force it using the newer version
# Also need to link all other *268 tools as well as riscv-openocd fails to build otherwise
mkdir -p autoconf_hack
if [ ! -f autoconf_hack/autoconf ]; then
        ln -s `which autoconf268` autoconf_hack/autoconf
        ln -s `which autoheader268` autoconf_hack/autoheader
        ln -s `which autom4te268` autoconf_hack/autom4te
        ln -s `which autoreconf268` autoconf_hack/autoreconf
        ln -s `which autoscan268` autoconf_hack/autoscan
        ln -s `which autoupdate268` autoconf_hack/autoupdate
fi
export PATH="`pwd`/autoconf_hack:$PATH"
```

I also had to locally build a new version of pkgconfig (0.29.2) to make things work. When running aclocal, autoconf and autoheader you'll have to pass the include folders of autoconf268 and pkg-config, in my case:
```
aclocal -I /usr/share/autoconf268/autoconf/ -I /tools/pkgconfig/install/pkg-config-0.29.2/share/aclocal/
autoconf -I /usr/share/autoconf268/autoconf/ -I /tools/pkgconfig/install/pkg-config-0.29.2/share/aclocal/
autoheader -I /usr/share/autoconf268/autoconf/ -I /tools/pkgconfig/install/pkg-config-0.29.2/share/aclocal/
```

Hope this is helpful to anyone trying to build riscv-openocd on Centos6...

KingFrige, 2020-07-30:
This is the solution: https://developer.aliyun.com/article/243363

check aclocal path:
```
 aclocal --print-ac-dir
```

copy  *.m4  to aclocal dir(aclocal --print-ac-dir, here is `/usr/local/share/aclocal`)
```
cp -rf /usr/share/aclocal/* /usr/local/share/aclocal
```