riscv-collab/riscv-openocd #142: Is it necessary to abort() if hart can't be halted?

mwachs5, 2017-11-30:
It may be impossible to halt a hart at a certain time. For example, a microarchitecture may allow blocking memory accesses relying on something external to the hart, or the hart may be in reset, or something. Currently, if this situation is encountered by the user, the code aborts:

https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L1803

Would it be possible to simply print a message in this case, and request the user to retry? Or ...? Anything that just kills the entire OpenOCD process isn't too helpful. I"m not sure what the implications would be to simply report failure and allow the user to try again.

palmer-dabbelt, 2017-12-01:
Every abort() in riscv-013.c is a bug, they're usually just there because I couldn't figure out how to actually get the error back up to the user without having everything fall apart.  It should be possible, do you mind plumbing the errors through and trying it out?

mwachs5, 2017-12-01:
Maybe the simplest thing (suggested by @p12nGH ) is to replace the hard-coded 256 loop here: 

https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L1792

with an infinite or user-specifiable timeout loop.

p12nGH, 2017-12-01:
I would make this loop infinite and add a small sleep (like 0.1s) between polls.

timsifive, 2017-12-01:
Infinite loops are just as bad as abort(). Every loop must have a timeout
of some kind.

mwachs5, 2017-12-01:
Yes, let's just make it a user-configurable timeout. It may cause problems with GDB if it is too long though (because if it is > than GDB's timeout, GDB may get confused). 

But in general, if that timeout fails, MUST we kill all of OpenOCD? Can we send some sort of RETRY error back or something?

timsifive, 2017-12-01:
I agree with Palmer: "Every abort() in riscv-013.c is a bug"

closed 2018-01-19 by timsifive