riscv-collab/riscv-openocd #157: GDB crashes with the register view

mwachs5, 2017-12-22:
I'm not sure if this is a GDB problem or an OpenOCD problem, but it is related to the #149 XML change.

In Eclipse, no registers show up anymore when using this updated version of OpenOCD. And, GDB crashes soon thereafter. Here is the log file:

```
Debug: 36537 1829 riscv-011.c:954 cache_write(): exit
Debug: 36538 1829 breakpoints.c:307 breakpoint_free(): free BPID: 0 --> 0
Debug: 36539 1830 gdb_server.c:2844 gdb_input_inner(): received packet: 'g'
Debug: 36540 1830 riscv.c:752 riscv_get_gdb_reg_list(): reg_class=1
Debug: 36541 1830 riscv.c:753 riscv_get_gdb_reg_list(): rtos_hartid=0 current_hartid=0
Debug: 36542 2161 gdb_server.c:2844 gdb_input_inner(): received packet: 'qXfer:threads:read::0,fff'
Debug: 36543 2161 gdb_server.c:2844 gdb_input_inner(): received packet: 'qXfer:threads:read::0,fff'
Debug: 36544 2598 gdb_server.c:2844 gdb_input_inner(): received packet: 'qXfer:threads:read::0,fff'
Debug: 36545 15580 gdb_server.c:1022 gdb_connection_closed(): GDB Close, Target: riscv.cpu, state: halted, gdb_actual_connections=0
Debug: 36546 15580 target.c:1524 target_call_event_callbacks(): target event 6 (gdb-end)
Debug: 36547 15580 target.c:1524 target_call_event_callbacks(): target event 24 (gdb-detach)
Info : 36548 15580 server.c:527 server_loop(): dropped 'gdb' connection
Debug: 36549 15582 riscv.c:264 riscv_deinit_target(): riscv_deinit_target()
Debug: 36550 15582 riscv-011.c:1418 deinit_target(): riscv_deinit_target()
```

Do I need to update the version of GDB to handle the XML file?

mwachs5, 2017-12-22:
Different weird output on v13 view... why do so few registers have values associated with them?
```
===== RISC-V Registers
(0) zero (/64)
(1) ra (/64)
(2) sp (/64)
(3) gp (/64)
(4) tp (/64)
(5) t0 (/64)
(6) t1 (/64)
(7) t2 (/64)
(8) fp (/64)
(9) s1 (/64)
(10) a0 (/64): 0x0000000000000000
(11) a1 (/64): 0x0000000000000000
(12) a2 (/64)
(13) a3 (/64)
(14) a4 (/64)
(15) a5 (/64)
(16) a6 (/64)
(17) a7 (/64)
(18) s2 (/64)
(19) s3 (/64)
(20) s4 (/64)
(21) s5 (/64)
(22) s6 (/64)
(23) s7 (/64)
(24) s8 (/64)
(25) s9 (/64)
(26) s10 (/64)
(27) s11 (/64)
(28) t3 (/64)
(29) t4 (/64)
(30) t5 (/64)
(31) t6 (/64)
(32) pc (/64): 0x0000000040400000
(833) mstatus (/64): 0x0000000200001800
(834) misa (/64)
(835) medeleg (/64)
(836) mideleg (/64)
(837) mie (/64)
(838) mtvec (/64)
(839) mcounteren (/64)
(868) mhpmevent3 (/64)
(869) mhpmevent4 (/64)
(870) mhpmevent5 (/64)
(871) mhpmevent6 (/64)
(872) mhpmevent7 (/64)
(873) mhpmevent8 (/64)
(874) mhpmevent9 (/64)
(875) mhpmevent10 (/64)
(876) mhpmevent11 (/64)
(877) mhpmevent12 (/64)
(878) mhpmevent13 (/64)
(879) mhpmevent14 (/64)
(880) mhpmevent15 (/64)
(881) mhpmevent16 (/64)
(882) mhpmevent17 (/64)
(883) mhpmevent18 (/64)
(884) mhpmevent19 (/64)
(885) mhpmevent20 (/64)
(886) mhpmevent21 (/64)
(887) mhpmevent22 (/64)
(888) mhpmevent23 (/64)
(889) mhpmevent24 (/64)
(890) mhpmevent25 (/64)
(891) mhpmevent26 (/64)
(892) mhpmevent27 (/64)
(893) mhpmevent28 (/64)
(894) mhpmevent29 (/64)
(895) mhpmevent30 (/64)
(896) mhpmevent31 (/64)
(897) mscratch (/64)
(898) mepc (/64)
(899) mcause (/64)
(900) mtval (/64)
(901) mip (/64)
(993) pmpcfg0 (/64)
(994) pmpcfg1 (/64)
(995) pmpcfg2 (/64)
(996) pmpcfg3 (/64)
(1009) pmpaddr0 (/64)
(1010) pmpaddr1 (/64)
(1011) pmpaddr2 (/64)
(1012) pmpaddr3 (/64)
(1013) pmpaddr4 (/64)
(1014) pmpaddr5 (/64)
(1015) pmpaddr6 (/64)
(1016) pmpaddr7 (/64)
(1017) pmpaddr8 (/64)
(1018) pmpaddr9 (/64)
(1019) pmpaddr10 (/64)
(1020) pmpaddr11 (/64)
(1021) pmpaddr12 (/64)
(1022) pmpaddr13 (/64)
(1023) pmpaddr14 (/64)
(1024) pmpaddr15 (/64)
(2017) tselect (/64)
(2018) tdata1 (/64)
(2019) tdata2 (/64)
(2020) tdata3 (/64)
(2033) dcsr (/64)
(2034) dpc (/64)
(2035) dscratch (/64)
(2881) mcycle (/64)
(2883) minstret (/64)
(2884) mhpmcounter3 (/64)
(2885) mhpmcounter4 (/64)
(2886) mhpmcounter5 (/64)
(2887) mhpmcounter6 (/64)
(2888) mhpmcounter7 (/64)
(2889) mhpmcounter8 (/64)
(2890) mhpmcounter9 (/64)
(2891) mhpmcounter10 (/64)
(2892) mhpmcounter11 (/64)
(2893) mhpmcounter12 (/64)
(2894) mhpmcounter13 (/64)
(2895) mhpmcounter14 (/64)
(2896) mhpmcounter15 (/64)
(2897) mhpmcounter16 (/64)
(2898) mhpmcounter17 (/64)
(2899) mhpmcounter18 (/64)
(2900) mhpmcounter19 (/64)
(2901) mhpmcounter20 (/64)
(2902) mhpmcounter21 (/64)
(2903) mhpmcounter22 (/64)
(2904) mhpmcounter23 (/64)
(2905) mhpmcounter24 (/64)
(2906) mhpmcounter25 (/64)
(2907) mhpmcounter26 (/64)
(2908) mhpmcounter27 (/64)
(2909) mhpmcounter28 (/64)
(2910) mhpmcounter29 (/64)
(2911) mhpmcounter30 (/64)
(2912) mhpmcounter31 (/64)
(3009) mcycleh (/64)
(3011) minstreth (/64)
(3012) mhpmcounter3h (/64)
(3013) mhpmcounter4h (/64)
(3014) mhpmcounter5h (/64)
(3015) mhpmcounter6h (/64)
(3016) mhpmcounter7h (/64)
(3017) mhpmcounter8h (/64)
(3018) mhpmcounter9h (/64)
(3019) mhpmcounter10h (/64)
(3020) mhpmcounter11h (/64)
(3021) mhpmcounter12h (/64)
(3022) mhpmcounter13h (/64)
(3023) mhpmcounter14h (/64)
(3024) mhpmcounter15h (/64)
(3025) mhpmcounter16h (/64)
(3026) mhpmcounter17h (/64)
(3027) mhpmcounter18h (/64)
(3028) mhpmcounter19h (/64)
(3029) mhpmcounter20h (/64)
(3030) mhpmcounter21h (/64)
(3031) mhpmcounter22h (/64)
(3032) mhpmcounter23h (/64)
(3033) mhpmcounter24h (/64)
(3034) mhpmcounter25h (/64)
(3035) mhpmcounter26h (/64)
(3036) mhpmcounter27h (/64)
(3037) mhpmcounter28h (/64)
(3038) mhpmcounter29h (/64)
(3039) mhpmcounter30h (/64)
(3040) mhpmcounter31h (/64)
(3137) cycle (/64)
(3138) time (/64)
(3139) instret (/64)
(3140) hpmcounter3 (/64)
(3141) hpmcounter4 (/64)
(3142) hpmcounter5 (/64)
(3143) hpmcounter6 (/64)
(3144) hpmcounter7 (/64)
(3145) hpmcounter8 (/64)
(3146) hpmcounter9 (/64)
(3147) hpmcounter10 (/64)
(3148) hpmcounter11 (/64)
(3149) hpmcounter12 (/64)
(3150) hpmcounter13 (/64)
(3151) hpmcounter14 (/64)
(3152) hpmcounter15 (/64)
(3153) hpmcounter16 (/64)
(3154) hpmcounter17 (/64)
(3155) hpmcounter18 (/64)
(3156) hpmcounter19 (/64)
(3157) hpmcounter20 (/64)
(3158) hpmcounter21 (/64)
(3159) hpmcounter22 (/64)
(3160) hpmcounter23 (/64)
(3161) hpmcounter24 (/64)
(3162) hpmcounter25 (/64)
(3163) hpmcounter26 (/64)
(3164) hpmcounter27 (/64)
(3165) hpmcounter28 (/64)
(3166) hpmcounter29 (/64)
(3167) hpmcounter30 (/64)
(3168) hpmcounter31 (/64)
(3265) cycleh (/64)
(3266) timeh (/64)
(3267) instreth (/64)
(3268) hpmcounter3h (/64)
(3269) hpmcounter4h (/64)
(3270) hpmcounter5h (/64)
(3271) hpmcounter6h (/64)
(3272) hpmcounter7h (/64)
(3273) hpmcounter8h (/64)
(3274) hpmcounter9h (/64)
(3275) hpmcounter10h (/64)
(3276) hpmcounter11h (/64)
(3277) hpmcounter12h (/64)
(3278) hpmcounter13h (/64)
(3279) hpmcounter14h (/64)
(3280) hpmcounter15h (/64)
(3281) hpmcounter16h (/64)
(3282) hpmcounter17h (/64)
(3283) hpmcounter18h (/64)
(3284) hpmcounter19h (/64)
(3285) hpmcounter20h (/64)
(3286) hpmcounter21h (/64)
(3287) hpmcounter22h (/64)
(3288) hpmcounter23h (/64)
(3289) hpmcounter24h (/64)
(3290) hpmcounter25h (/64)
(3291) hpmcounter26h (/64)
(3292) hpmcounter27h (/64)
(3293) hpmcounter28h (/64)
(3294) hpmcounter29h (/64)
(3295) hpmcounter30h (/64)
(3296) hpmcounter31h (/64)
(3922) mvendorid (/64)
(3923) marchid (/64)
(3924) mimpid (/64)
(3925) mhartid (/64)
(4161) priv (/8)
Info : dropped 'gdb' connection
```

timsifive, 2017-12-22:
Can you clarify what exact gdb/OpenOCD versions you are using?

ISTR a new gdb is required, but I don't recall the details.

timsifive, 2017-12-22:
New OpenOCD (6c719f0ab838e6804500fa8ac6917b34a78ecf3e) with old gdb (b69b760d69eab4f9d9d761da38c4275d88a1d94c) mostly works. The only oddity I noticed is that FPRs show up as existing, but they don't have names. Them not having names might throw off Eclipse.

I wouldn't call this a supported combination of versions to use.

mwachs5, 2017-12-22:
I was trying with GDB from riscv64-unknown-elf-gcc-20170612 (https://github.com/riscv/riscv-gnu-toolchain/releases/tag/v20170612). Which is pretty ancient I'll admit. I'll try the current GDB master

closed 2018-03-03 by timsifive