riscv-collab/riscv-openocd #158: -coreid x instead of -rtos riscv?

TM1234, 2017-12-22:
At the moment the RISC-V openocd support allows for two modes of target declaration:

```
set _TARGETNAME $DEVICE.cpu
target create $_TARGETNAME riscv -chain-position $DEVICE.dap -coreid 0
```

and

```
set _TARGETNAME $DEVICE.cpu
target create $_TARGETNAME riscv -chain-position $DEVICE.dap -rtos riscv
```

The former creates a debug connection to a specified core/hart.
The latter creates a single debug connection to a target with possibly more than one core/hart and treats the harts as if they were software threads while debugging.

Would it not be better to use something like:

```
set _TARGETNAME $DEVICE.cpu
target create $_TARGETNAME riscv -chain-position $DEVICE.dap -coreid -1
```

or 

```
set _TARGETNAME $DEVICE.cpu
target create $_TARGETNAME riscv -chain-position $DEVICE.dap -coreid 0xffffffff[ffffffff]
```

or something similar for the second second scenario thereby not precluding the use of the original -rtos <rtos-type> option in openocd for RTOS debugging?

The current approach using -rtos riscv hijacks the -rtos option for non standard purposes.

(Another possibility might be to number harts from 1 rather than 0 and then use 0 as the sentinel value to indicated the behaviour currently imposed by -rtos riscv. 0 based numbering has the advantage of aligning with how gdb names/numbers the "threads").

timsifive, 2017-12-22:
Overloading -rtos to debug multiple harts with one OpenOCD is a hack that is not easily fixed. It goes much deeper than simply the syntax you describe here.

It would be great to get rid of it, but as it is people apparently really like the gdb user experience where gdb simply sees harts as threads. Achieving this without overriding -rtos is a big project.

mwachs5, 2017-12-22:
Does the RTOS flag usually apply in the case where there are actually
multiple hardware threads? It seems like the two are generally mutually
exclusive?

timsifive, 2017-12-22:
I'm not sure there is a "usually." gdb does not have a concept of cores at all. I can imagine a debugger that is aware of multiple cores as well as threads under an RTOS, that allows you to debug eg. some kernel code, while also being aware of and debugging individual threads.

TM1234, 2018-01-30:
Since the use of -rtos riscv is unlikely to be changed at this stage perhaps I should simply close this issue?
It may be that multi-core/hart targets are unlikely to be running the lightweight RTOSs (FreeRTOS etc.) and will probably be running Linux?
And I suppose a single core/hart debug connection using -rtos FreeRTOS (or whatever) can be used if needed?