riscv-collab/riscv-openocd #175: For known cores, only expose those CSRs we know they support

timsifive, 2018-01-04:
We can recognize cores by the JTAG IDCODE or by reading mvendorid/marchid.

ilg-ul, 2018-01-04:
Perhaps you should add a folder with XML files for various JTAG IDCODEs and mvendorid/marchid.

You can encode the IDs as hex digits in the file name and automatically pass the XML to GDB if you can match a file.

mwachs5, 2018-01-04:
Do you already have a list in mind? I don't know of any useful differentiation at the moment or anything planned to be encoded in the IDCODE (speaking for SiFive, not anything general).

timsifive, 2018-01-04:
A list of cores, or registers?

People are annoyed that eg. non-existent performance counters show up when debugging HiFive1.

mwachs5, 2018-01-04:
I meant cores.... this seems like a good idea but not likely to get far in practice (again speaking for SiFive case... knowing the IDCODE  + mvendor + marchid is not going to tell you much about how many perf. counters there are). I think in general we should put pressure on ISA spec to make those things discoverable

timsifive, 2018-01-04:
IDCODE has to be changed (per spec) if a different BSDL is required to test the device. While it's possible to change CSRs while keeping the BSDL the same, it's a pretty good heuristic.

"The combination of mvendorid and marchid should uniquely identify the type of hart microarchitecture that is implemented." I'm not clear whether the list of CSRs is part of the microarchitecture, but at least this is also a good heuristic.

Is there something better to look at?

I agree that the ISA spec should be updated for these things to be discoverable, but currently they are not.

ilg-ul, 2018-01-04:
> I don't know of any useful differentiation at the moment or anything planned to be encoded in the IDCODE

I'm not sure I understand what you mean.

as far as I know, IDCODE is used to include a manufacturer code (derived from the JEDEC Standard Manufacturer's Identification Code standard, JEP-106), a part number assigned by the manufacturer, and a part version code. 

so, if you have unique IDCODEs, you indirectly know everything about the actual device, down to part version. 

in other words, you define an XML file with the device registers and find a method to identify the file based on IDCODE.

mwachs5, 2018-01-04:
That may very well be correct usage of IDCODE, but if you need to rely on IDCODE to figure out some detail about the RISC-V core architecture, there is a problem with the RISC-V ISA (or perhaps config string) spec. Lots of software may be interested in the presence of performance counters (to use your example) and that software is not going to have access to the IDCODE.

mvendorid/marchid are allowed to be not implemented => 0

Anyway, I'm not trying to argue that you shouldn't do this, just saying it may not be that helpful. The RISC-V debug spec was written assuming you could discover what you wanted to know rather than having to rely on someone telling you externally (and needing the IDCODE -> XML mapping seems like someone telling you externally).

mwachs5, 2018-01-04:
Sorry, why can't you just discover the CSRs that are/are not present by trying to read them and seeing if you get exception...? Is that just too slow?

timsifive, 2018-01-04:
Reading CSRs may have side effects.

timsifive, 2021-01-15:
We do a decent job detecting available CSRs. Maybe we'll add this feature but I haven't seen any interest.

closed 2021-01-15 by timsifive