riscv-collab/riscv-openocd #177: data-list-register-names should return only existing registers, not a huge array with empty names

ilg-ul, 2018-01-04:
as already discussed in other issues, huge responses to the MI `data-list-register-names` command  breaks some debugger environments, like Eclipse, so the list of registers returned by this command should be kept to a minimum.

the current implementation places the CSR names in locations based on the CSR numbers, with many empty locations.

a better strategy would be to compact this long array and keep an internal translation table.

timsifive, 2018-01-05:
I thought that we had gotten under whatever the limit was for Eclipse crashing a long time ago. Does it still crash?

ilg-ul, 2018-01-05:
we do not know what the limit is, if it is static, or whatever the problem actually is.

all we know is that unreasonably large responses to `data-list-register-names` kill Eclipse.

does the current response seem reasonable to you?

```
642,219 47-data-list-register-names --thread-group i1
642,226 47^done,register-names=["zero","ra","sp","gp","tp","t0","t1","t2","fp","s1","a0","a1","a2","\
a3","a4","a5","a6","a7","s2","s3","s4","s5","s6","s7","s8","s9","s10","s11","t3","t4","t5","t6","pc"\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"mstatus","misa","medeleg","mideleg","mie","mtvec","mcounteren","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","mhpmevent3","mhpmevent4","mhpmevent5","mhpmevent6"\
,"mhpmevent7","mhpmevent8","mhpmevent9","mhpmevent10","mhpmevent11","mhpmevent12","mhpmevent13","mhp\
mevent14","mhpmevent15","mhpmevent16","mhpmevent17","mhpmevent18","mhpmevent19","mhpmevent20","mhpme\
vent21","mhpmevent22","mhpmevent23","mhpmevent24","mhpmevent25","mhpmevent26","mhpmevent27","mhpmeve\
nt28","mhpmevent29","mhpmevent30","mhpmevent31","mscratch","mepc","mcause","","mip","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","pmpcfg0","pmpcfg1","pmpcfg2","pmpcfg3","",\
"","","","","","","","","","","","pmpaddr0","pmpaddr1","pmpaddr2","pmpaddr3","pmpaddr4","pmpaddr5","\
pmpaddr6","pmpaddr7","pmpaddr8","pmpaddr9","pmpaddr10","pmpaddr11","pmpaddr12","pmpaddr13","pmpaddr1\
4","pmpaddr15","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
[... cut ...]
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",\
"","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","\
","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","",""\
,"","","","","","","","","","","","","priv","mtval"]
642,230 (gdb) 
```

timsifive, 2018-01-05:
The tradeoff is more complex code in gdb, or a more verbose register list in an API. I pick the latter.

timsifive, 2018-01-29:
I'm not planning to address this, and it's a gdb issue in any case.

closed 2018-01-29 by timsifive

ilg-ul, 2018-01-29:
It is a GDB issue because OpenOCD does not pass an XML with a shorter list of actual registers.