riscv-collab/riscv-openocd #182: compilation error on macOS

mwachs5, 2018-01-09:
Not sure if this correlates to a recent change, but current OpenOCD master I cannot build on macOS:

I think this is related to bumping libjaylink submodule.

```
discovery_tcp.c:305:9: error: implicit declaration of function 'select' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
                ret = select(sock + 1, &rfds, NULL, NULL, &timeout);

```

timsifive, 2018-01-09:
That looks busted in OpenOCD. I would suggest to do MacOS by with `./configure --disable-werror`.

mwachs5, 2018-01-09:
never mind. I just started from a clean checkout and it worked. Must have been something stale

closed 2018-01-09 by mwachs5

codebeautiful, 2018-12-24:
well, i complied successful
`
./bootstrap
./configure CFLAGS=--std=gnu11
make -j8
sudo make install
`

filamoon, 2019-05-22:
I got the exactly same error on mac and none of the above helped. Please help...