riscv-collab/riscv-openocd #185: Add semihosting support

ilg-ul, 2018-01-15:
With the semihosting call more or less defined, and the Eclipse template already generating projects using semihosting for the trace messages, the next step is to implement it in OpenOCD.

The current workaround to compensate for the lack of multiple EBREAKs is:

```asm
 .option norvc 
 slli x0, x0, 0x1f
 ebreak
 srai x0, x0, 7
```

If you agree, I'll first experiment a solution in the GME OpenOCD fork first, then contribute it back.

My plan is to do it in several steps:

- make the current semihosting implementation more generic and move it outside the ARM folder
- modify the ARM code to use the new implementation; this will (more or less) validate the changes
- add code to the RISC-V folders to identify the use the new file

In the first tests I'll use only the calls required to pass trace messages via the debug channel.

In a later phase I'll test the other calls, to pass the command line, to return an exit code, to open/read/write/close files, etc.

Any suggestions?

sorear, 2018-01-16:
Has the set of semihosting call numbers been defined yet?

ilg-ul, 2018-01-16:
The `do_semihosting(struct target *target)` uses the pointer to access target specific info.

@timsifive, what is the recommended way to identify the RV32 vs RV64 targets? 

As far as I can see, the current semihosting code supports only 32-bits target, and I need to upgrade it.

ilg-ul, 2018-01-16:
> semihosting call numbers

yes, they were defined by ARM long time ago.

TM1234, 2018-01-16:
> @timsifive, what is the recommended way to identify the RV32 vs RV64 targets?

Isn't it via the misa CSR?

sorear, 2018-01-16:
QEMU uses completely different call numbers for semihost operations on arm, mips, xtensa, m68k, and lm32, so while copying the arm table to start is a viable approach, it's not the only one.

One of my goals for this project is to replace the [Berkeley fesvr semihosting](https://github.com/riscv/riscv-fesvr/blob/master/fesvr/syscall.cc#L54-L74) with a standards-track mechanism; this requires most of the usual suspects as well as a non-blocking tty input call.

ilg-ul, 2018-01-16:
> copying the arm table to start is a viable approach, it's not the only one.

I do not know the reasons why other platforms use different numbers, but, given that ARM invented this, I would preserve compatibility with the original specifications, and if other calls are needed, I would add them as extentions.

timsifive, 2018-01-16:
On Mon, Jan 15, 2018 at 3:02 PM, Liviu Ionescu <notifications@github.com>
wrote:

> If you agree, I'll first experiment a solution in the GME OpenOCD fork
> first, then contribute it back.
>
> My plan is to do it in several steps:
>
>    - make the current semihosting implementation more [...]
This sounds like a good plan.

Tim

timsifive, 2018-01-16:
On Tue, Jan 16, 2018 at 5:54 AM, Liviu Ionescu <notifications@github.com>
wrote:

> The do_semihosting(struct target *target) uses the pointer to access
> target specific info.
>
> @timsifive <https://github.com/timsifive>, what is the recommended way to
> identify the RV32 vs RV64 targets?
>
I assume you mean from OpenOCD code. If you know the target is a RISC-V
target, you can call riscv_xlen(target). If you want to generally know
whether it's a 64-bit target, I'm not sure. It may be safe to look at the
first register in target->reg_cache and check its size. Even better,
presumably you know the names of the registers used to pass arguments. You
can look them up with register_get_by_name() and get their size.

Tim

ilg-ul, 2018-01-16:
> riscv_xlen(target)

thanks, this should do it.

ilg-ul, 2018-01-16:
@timsifive, I prepared my development environment, I added code to process the new commands to enable semihosting, to get the command line, etc. and now I would need a bit of help to find the proper place to insert the code to test the semihosting condition.

For ARM, the function `arm_semihosting()` is periodically called from `hla_target.c:adapter_poll()`, which is reached via `target->type->poll(target)`.

For RISC-V, on the `->poll` callback we have `old_or_new_riscv_poll()`, which dispatches to `oldriscv_poll()` or to `riscv_openocd_poll()`.

`oldriscv_poll()` calls `->poll` again (which confuses me completely).

`riscv_openocd_poll()` does some magic that I do not understand, but mainly calls one or multiple `riscv_poll_hart()` and in some conditions decides that the target is halted.

Personally I would call the `riscv_semihosting()` somewhere at the end of `riscv_openocd_poll()`, after it is obvious that the target is halted and the reason of the halt is a breakpoint, somewhere right before the `target_call_event_callbacks()` at the end.

This is somehow consistent with the ARM usage, (https://github.com/riscv/riscv-openocd/blob/6f9585068f3d3a6e94d89406051412d12c9c22e9/src/target/hla_target.c#L455-L473).

What do you think?

ilg-ul, 2018-01-17:
Unfortunatelly on my HiFive1, the `riscv_openocd_poll()` is not called when an EBREAK is entered, other functions like `riscv-011.c:handle_halt()` are used.

I'll experiment and find a location where the semihosting test can be placed, but I do not have a good understanding of the current code, so I need your help to decide on the final location.

timsifive, 2018-01-17:
The 0.11/0.13 abstract layer is pretty messy. Cleaning it up is somewhere
on my todo list.

The simplest solution is probably to modify 0.11 like you did, and also
modify riscv_openocd_poll(). Note that in the latter, depending on whether
-rtos is used or not, there may be more than one hart and conceivably any
of them can halt for a semihosting call at any time. If you need an 0.13
target, you can use spike with --rbb-port.

Tim
​

TM1234, 2018-01-17:
>  If you need an 0.13 target, you can use spike with --rbb-port.

I don't understand this - why would one not add semihosting to 0.13 too?

ilg-ul, 2018-01-19:
The first version of the functional semihosting is on GitHub.

https://github.com/gnu-mcu-eclipse/openocd/compare/gnu-mcu-eclipse-dev...semihosting

Briefly, I added:

- target/semihosting_common.h
- target/semihosting_common.c
- target/riscv/riscv_semihosting.c

As for changes affecting the common files, I only added a new member `semihosting` to the `struct target`.

The integration into the RISC-V code requires a call to initialize the subsystem and calls from the target poll routines.

For 0.13, things looked more easy to understand, I used the function `riscv_openocd_poll()`, similarly to the ARM code.

For 0.11 the code is less structured, and I could not find an equivalent function, so I used `handle_halt()`, which seems functional, but I'm not sure it is the right place.

My tests currently exercise only the `trace::printf()` functionality. Both E51Arty and HiFive1 were functional.

However, on HiFive1 I occasionaly noticed timeouts, which make me think that either the code requires some cleaning, or we might try another location for the semihosting call.

Next week I'll update the template project to generate full semihosted applications, and test some of the semihosting syscalls.

ilg-ul, 2018-01-23:
GNU MCU Eclipse with full semihosting support is out:

https://gnu-mcu-eclipse.github.io/blog/2018/01/23/openocd-v0-10-0-7-20180123-released/

Both 32/64-bits targets are supported.

The main implementation files are:

- https://github.com/gnu-mcu-eclipse/openocd/blob/gnu-mcu-eclipse-dev/src/target/semihosting_common.c
- https://github.com/gnu-mcu-eclipse/openocd/blob/gnu-mcu-eclipse-dev/src/target/riscv/riscv_semihosting.c

I'll use it in my projects for a while, and fix possible bugs, then contribute it back.

TM1234, 2018-01-23:
Hi Liviu - is there an implementation of the RISC-V target side semihosting syscall code available somewhere?
Thanks

ilg-ul, 2018-01-23:
The main target implementation file is: 

https://github.com/micro-os-plus/semihosting-xpack/blob/xpack/src/syscalls-semihosting.cpp

The C++ implementation is relatively clean. The C implementation is less clean, since there is a small trick here, due to newlib specifics, some of the functions are aliased to underscore functions, some to non-underscore functions.

At the end are some functions required by the portable µOS++ startup, for example to process argc/argv and to terminate an application.

The RISC-V semihostign call is in:

https://github.com/micro-os-plus/riscv-arch-xpack/blob/xpack/include/riscv-arch/arch-semihosting-inlines.h

All these files are used by the latest version of the Eclipse SiFive template, to be released probably tomorrow.

TM1234, 2018-01-24:
Just curious (and it's not urgent) do you recall what date/hash you forked the sifive code (https://github.com/riscv/riscv-openocd) as there seem to be quite a few changes to the sifive code (particularly in riscv register handling) compared to your code.

ilg-ul, 2018-01-24:
The last time I merged this repo was Jan 18th, cb543e804d88ae2421292bc2d85185e9dc7cc5b5.

TM1234, 2018-01-24:
Thanks - strange that changes such as these dated earlier (Jan 8) are not included:

https://github.com/riscv/riscv-openocd/commit/160b79427480c1a7d26bda0c45d8016ec2471b58

ilg-ul, 2018-01-25:
Oops! You're right, it looks like the last actual merge was on Dec 28th, so I missed some of the recent commits.

@timsifive, when do you plan to update to the latest upstream commits, so I can plan a new release?

TM1234, 2018-01-25:
Thanks Liviu.
I wasn't sure if I had misunderstood the chronology/mechanics of the commits to the main/sifive repo.
So you take things from the main OpenOCD project rather than the riscv/sifive repo?

ilg-ul, 2018-01-25:
> you take things from the main OpenOCD project rather than the riscv/sifive repo

it depends how often the riscv is updated.

I'd prefer the riscv repo to be more or less up to date, and follow only the riscv repo.

TM1234, 2018-01-25:
OK - thanks Liviu.

timsifive, 2018-01-25:
On Wed, Jan 24, 2018 at 4:07 PM, Liviu Ionescu <notifications@github.com>
wrote:

> @timsifive <https://github.com/timsifive>, when do you plan to update to
> the latest upstream commits, so I can plan a new release?
>
In the next several days.

Tim

ilg-ul, 2018-01-25:
great, thank you, Tim.

TM1234, 2018-02-12:
Is this the "update to latest upstream commits" referred to above?

https://github.com/riscv/riscv-openocd/commit/6cbb45f7f1eb04470f69472917cb986a238deee4

timsifive, 2018-11-07:
This was done by #266, right?

ilg-ul, 2018-11-07:
I guess it was done.

closed 2018-11-07 by ilg-ul