riscv-collab/riscv-openocd #190: [DSA 4093-1] openocd security update

palmer-dabbelt, 2018-01-23:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=887488

timsifive, 2018-01-23:
FWIW, discussion on the mailing list indicates the fix addresses this one particular attack vector, but there are likely many others.

We'll get this fix next time we merge from upstream, which I was planning to do once I get http://openocd.zylin.com/#/c/4312/ merged. I can hurry this up if you think it's important.

timsifive, 2018-02-05:
It looks like this hasn't actually merged to OpenOCD mainline yet, so we're not getting a fix in PR #205.

timsifive, 2018-05-16:
OpenOCD mailing list discussion: https://sourceforge.net/p/openocd/mailman/openocd-devel/thread/270e864949d7dcbac4695d42d2b9bba5%40atx.name/
Proposed fix: http://openocd.zylin.com/#/c/4335/

If/when this merges to mainline OpenOCD, we'll get it when we pull in mainline. I don't think this requires separate action on our part.

closed 2018-05-16 by timsifive