riscv-collab/riscv-openocd #193: Supress polling messages with `-d` on

mwachs5, 2018-01-24:
If I am running with `-d`, anything interesting (say, accessing a memory location with GDB that comes back with an error) is swamped with the debugging output from the "poll" messages. Is it possible to suppress those poll messages even when `-d` flag is used?

(a subpart of #46)

timsifive, 2018-01-25:
Yes, but I like the simplicity of telling somebody to run "openocd -d" to get a log which contains everything that actually happened.

I'm open to suppressing poll messages with a more complex command line or something. I'm not sure if OpenOCD has any more flags that can be easily used, though. How about some conditional compilation?

mwachs5, 2018-01-25:
How about a risc-v specific command?

TM1234, 2018-01-30:
-d implies -d 3.
Maybe -d n (n < 3) doesn't emit poll messages?
I haven't tried it myself.
From the openocd user's guide:

http://openocd.org/doc/html/General-Commands.html

> Command: debug_level [n]
> Display debug level. If n (from 0..4) is provided, then set it to that level. This affects the kind of messages sent to the server log. Level 0 is error messages only; level 1 adds warnings; level 2 adds informational messages; level 3 adds debugging messages; and level 4 adds verbose low-level debug messages. The default is level 2, but that can be overridden on the command line along with the location of that log file (which is normally the server’s standard output). See Running.

TM1234, 2018-01-30:
Curiously the above snippet says that the default logging level is 2 but this says 3.
RTFS to find out the truth I guess...

http://openocd.org/doc/html/Running.html#Running

```
--debug      | -d       set debug level to 3
             | -d<n>    set debug level to <level>
```

mwachs5, 2018-01-31:
Given that, perhaps we can make the polling messages under the "verbose" level?

timsifive, 2018-01-31:
It would be ideal if eg. -d2 could be used to show all debugging without
the polling messages, but unfortunately 2 is used for info. Here are all
the levels:

/* logging priorities
 * LOG_LVL_SILENT - turn off all output. In lieu of try + catch this
can be used as a
 *                  feeble ersatz.
 * LOG_LVL_USER - user messages. Could be anything from information
 *                to progress messags. These messages do not represent
 *                incorrect or unexpected behaviour, just normal execution.
 * LOG_LVL_ERROR - fatal errors, that are likely to cause program abort
 * LOG_LVL_WARNING - non-fatal errors, that may be resolved later
 * LOG_LVL_INFO - state information, etc.
 * LOG_LVL_DEBUG - debug statements, execution trace
 * LOG_LVL_DEBUG_IO - verbose debug, low-level I/O trace
 */
enum log_levels {
    LOG_LVL_SILENT = -3,
    LOG_LVL_OUTPUT = -2,
    LOG_LVL_USER = -1,
    LOG_LVL_ERROR = 0,
    LOG_LVL_WARNING = 1,
    LOG_LVL_INFO = 2,
    LOG_LVL_DEBUG = 3,
    LOG_LVL_DEBUG_IO = 4,
};

We could only show polling on level 4, but then it won’t show up when users
simply run openocd -d which might hurt debugging of problems they’ve
reported. Megan’s suggestion of adding a RISC-V-specific command seems the
best solution.
​

closed 2021-01-15 by timsifive