riscv-collab/riscv-openocd #196: investigate output of `poll`

timsifive, 2018-01-26:
This is verbose on ARM. http://openocd.org/doc/html/Server-Configuration.html says eg.:
```
> poll
background polling: on
target state: halted
target halted in ARM state due to debug-request, \
               current mode: Supervisor
cpsr: 0x800000d3 pc: 0x11081bfc
MMU: disabled, D-Cache: disabled, I-Cache: enabled
>
```

On RISC-V we don't get that output:
```
> poll
background polling: on
TAP: riscv.cpu (enabled)
> status
```

timsifive, 2018-05-16:
I think the ARM example is old. Current mainline OpenOCD displays the `TAP: ...` line instead of `target state: ...` and that seems to have been the case since 2008. Then the remaining output comes from the target's `arch_state()` which is supposed to "Issue USER() w/architecture specific status." We could print out which hardware trigger was halted, but aside from that I can't imagine what we'd want to display.

TM1234, 2018-08-20:
No - a build of the very latest openocd from https://github.com/riscv/riscv-openocd gives the following output for a Cortex-M3 target poll:

```
Open On-Chip Debugger                                        
> poll                                                       
background polling: on                                       
TAP: M2S090.cpu (enabled)                                    
> reset init                                                 
target halted due to debug-request, current mode: Thread     
xPSR: 00000000 pc: 00000000 msp: 00000000                    
lr (/32): 0x00000000                                         
xPSR (/32): 0x01000000                                       
> poll                                                       
background polling: on                                       
TAP: M2S090.cpu (enabled)                                    
target halted due to debug-request, current mode: Thread     
xPSR: 0x01000000 pc: 00000000 msp: 00000000                  
>                                                         
```   
Note that the first poll gives abbreviated info but the one after reset init gives more info including the current target state. I'm not sure why the output is not consistent.

Note also that the documentation says this about the poll command (with no "on" or "off" argument):

> Poll the current target for its current state. (Also, see [target curstate], page 68.)
> If that target is in debug mode, architecture specific information about the current
> state is printed. An optional parameter allows background polling to be enabled and
> disabled.

So I believe that the RISC-V implementation of poll should act similarly and in accordance with the documentation.

en-sc, 2024-01-26:
The issue is addressed by #1002.

closed 2024-01-26 by en-sc