riscv-collab/riscv-openocd #202: After `flash bank` command, can't access 64-bit addresses

mwachs5, 2018-01-31:
If I have a target that supports > 32-bit addresses, OpenOCD generally works to access the memory regions > 32 bits. However, if I add a `flash bank` command in my `.cfg` script, GDB seems to no longer even attempt to access the upper addresses (there is no record in the OpenOCD log of the attempt to access them, GDB just says "can't access memory at <high location>". It may be a GDB issue, but I am suspecting that the XML memory map that OpenOCD passes (the same one that would indicate that GDB needs to use the appropriate commands when accessing memory-mapped flash or using `load` to program flash) is somehow misinforming GDB that the larger addresses are inaccessible.

timsifive, 2018-02-01:
Can you share the specific `flash bank` commands you used?

mwachs5, 2018-02-01:
`flash bank onboard_spi_flash fespi 0x20000000 0 0 0 $_TARGETNAME 0x10040000`

timsifive, 2018-02-01:
The problem is that OpenOCD tells gdb where RAM is, and (probably) uses 32-bit math to figure out the length of memory. Here's what OpenOCD tells gdb in this case:
```
Debug: 3878 1070 gdb_server.c:393 gdb_put_packet_inner(): sending packet '$l<memory-map>
<memory type="ram" start="0x0" length="0x20000000"/>
<memory type="ram" start="0x20000000" length="0xe0000000"/>
</memory-map>
#f6'
```

I'm digging into that code next.

timsifive, 2018-04-24:
Yes. OpenOCD flash bank code uses 32-bit addresses. From the OpenOCD mailing list (4/4/18):

> Sure we need to change base to target_addr_t and maybe introduce a target_size_t?
> Probably there is no nor flash sized over 4GB but possible overflow in 32-bit size arithmetic should be prevented. 
> I suppose the change would require some (lot of?) changes in the flash core and drivers too (printf formats...).
> Feel free to make a patch.
> 
> I changed some occurrences of the flash address in http://openocd.zylin.com/4399
> but this is far from complete, changes are just in touched code.
> There are also possibly conflicting changes http://openocd.zylin.com/4297
> and 4298 - I'm going to merge all of them soon.

So once we have those changes (we may already) the plan is to update this code to use 64-bit addresses.

timsifive, 2020-03-30:
Fixed.

closed 2020-03-30 by timsifive