riscv-collab/riscv-openocd #211: Debugging a hart not robust across non-debugger controlled reset

mwachs5, 2018-02-09:
Consider the case where openOCD is debugging a hart and one of its actions cause the hart to be reset (e.g writing a memory location which leads to soft reset), or someone presses an external reset button. OpenOCD doesn't re-halt the hart or otherwise gracefully recover (it works fine if OpenOCD knowingly caused the reset with `ndreset`).

Most of the problems could be avoided if OpenOCD left `haltreq` set until it actually wanted to resume. This would cause a hart which was being debugged and got reset to halt again even if it was reset out of the debugger's control.

mwachs5, 2018-03-16:
#226 doesn't help this. It just tells the user, but it doesn't actually keep the hart halted (I think, please correct me if I'm wrong).

timsifive, 2018-03-19:
You're right. It gets us closer, but doesn't implement what this issue asks for. I'll get to that.

closed 2018-05-01 by timsifive

mwachs5, 2018-05-01:
#226 doesn't resolve this issue as noted in your comment above (I thought)

timsifive, 2018-05-01:
Yes. I think github found a magic string in one of the #226 commits and automatically closed this issue.

elfmimi, 2020-04-09:
I believe dmstatus should be monitored, at least fetched again, after writing halt request to dmcontrol.
https://github.com/riscv/riscv-openocd/blob/cbb15587dc782ac8ade7ae252e7b760cfba4a178/src/target/riscv/riscv-013.c#L3852-L3856

PR #226 (2018-05-01): Notice when a hart is reset while it's being debugged, and let the user know that it happened
I think this resolves #211.