riscv-collab/riscv-openocd #229: Compare speed butween riscv-011 and riscv-013

Tong-Lin, 2018-03-22:
We compare reading a work from memory butween riscv-011 and riscv-013, riscv-011(15-20ms) is much faster than riscv-013(300-400ms); 

riscv-011 -d3 log:

> Debug: 777 6299 gdb_server.c:2850 gdb_input_inner(): received packet: 'm30000,4'
Debug: 778 6299 gdb_server.c:1416 gdb_read_memory_packet(): addr: 0x0000000000030000, len: 0x00000004
Debug: 779 6299 target.c:2097 target_read_buffer(): reading buffer of 4 byte at 0x00030000
Debug: 780 6299 riscv-011.c:753 cache_set32(): cache[0x0] = 0x41002403: DASM(0x41002403) (hit)
Debug: 781 6299 riscv-011.c:756 cache_set32(): cache[0x1] = 0x00042483: DASM(0x42483)
Debug: 782 6300 riscv-011.c:756 cache_set32(): cache[0x2] = 0x40902823: DASM(0x40902823)
Debug: 783 6300 riscv-011.c:756 cache_set32(): cache[0x3] = 0x3f80006f: DASM(0x3f80006f)
Debug: 784 6300 riscv-011.c:845 cache_write(): enter
Debug: 785 6300 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723575-508737
Debug: 786 6305 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723575-513211
Debug: 787 6305 riscv-011.c:408 scan(): 41b w .h:00042483 @01 -> + .h:41002403 @00
Debug: 788 6306 riscv-011.c:408 scan(): 41b w .h:40902823 @02 -> + .h:7a041073 @01
Debug: 789 6306 riscv-011.c:408 scan(): 41b w .h:3f80006f @03 -> + .h:3fc0006f @02
Debug: 790 6306 riscv-011.c:965 cache_write(): exit
Debug: 791 6306 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723575-514468
Debug: 792 6310 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723575-518720
Debug: 793 6310 riscv-011.c:408 scan(): 41b w ih:00030000 @04 -> + .h:3f80006f @03
Debug: 794 6310 riscv-011.c:408 scan(): 41b r .h:00000000 @04 -> + .h:00000000 @04
Debug: 795 6310 riscv-011.c:408 scan(): 41b r .h:00000000 @06 -> + .h:00f91d97 @04
Debug: 796 6310 riscv-011.c:408 scan(): 41b r .h:00000000 @06 -> + .h:00000000 @06
Debug: 797 6310 riscv-011.c:2105 read_memory(): j=0 status=0 data=13f80006f
Debug: 798 6310 riscv-011.c:2105 read_memory(): j=1 status=0 data=100000000
Debug: 799 6310 riscv-011.c:2105 read_memory(): j=2 status=0 data=100f91d97
Debug: 800 6310 riscv-011.c:2105 read_memory(): j=3 status=0 data=100000000

riscv-013 -d3 log:

> Debug: 4982 8526 gdb_server.c:2850 gdb_input_inner(): received packet: 'm30000,4'
Debug: 4983 8526 gdb_server.c:1416 gdb_read_memory_packet(): addr: 0x0000000000030000, len: 0x00000004
Debug: 4984 8526 target.c:2097 target_read_buffer(): reading buffer of 4 byte at 0x00030000
Debug: 4985 8526 riscv.c:1536 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 4986 8526 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-465818
Debug: 4987 8529 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-468468
Debug: 4988 8529 riscv-013.c:302 scan(): 41b r 00000000 @10 -> + 00000000 @00
Debug: 4989 8529 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-468614
Debug: 4990 8532 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-471981
Debug: 4991 8533 riscv-013.c:302 scan(): 41b - 00000000 @10 -> + 00000001 @10
Debug: 4992 8533 riscv-013.c:313 scan():  ->  dmactive
Debug: 4993 8533 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-472056
Debug: 4994 8535 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-474908
Debug: 4995 8535 riscv-013.c:302 scan(): 41b w 00000001 @10 -> + 00000000 @00
Debug: 4996 8535 riscv-013.c:313 scan():  dmactive -> 
Debug: 4997 8535 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-474961
Debug: 4998 8539 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-478104
Debug: 4999 8539 riscv-013.c:302 scan(): 41b - 00000000 @10 -> + 00000001 @10
Debug: 5000 8539 riscv-013.c:313 scan():  ->  dmactive
Debug: 5001 8539 riscv-013.c:1684 read_memory_progbuf(): reading 1 words of 4 bytes from 0x30000
Debug: 5002 8539 riscv-013.c:599 execute_abstract_command(): command=0x221008
Debug: 5003 8539 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-478216
Debug: 5004 8543 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-482158
Debug: 5005 8543 riscv-013.c:302 scan(): 41b w 00221008 @17 -> + 00000000 @00
Debug: 5006 8543 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-482201
Debug: 5007 8546 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-485404
Debug: 5008 8546 riscv-013.c:302 scan(): 41b - 00000000 @17 -> + 00231008 @17
Debug: 5009 8546 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-485448
Debug: 5010 8549 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-488969
Debug: 5011 8550 riscv-013.c:302 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 5012 8550 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-489013
Debug: 5013 8552 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-491969
Debug: 5014 8553 riscv-013.c:302 scan(): 41b - 00000000 @16 -> + 10000001 @16
Debug: 5015 8553 riscv-013.c:313 scan():  ->  progbufsize=16 datacount=1
Debug: 5016 8553 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-492041
Debug: 5017 8556 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-494995
Debug: 5018 8556 riscv-013.c:302 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 5019 8556 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-495037
Debug: 5020 8558 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-497957
[... 250 lines cut ...]
Debug: 5271 8823 riscv-013.c:302 scan(): 41b w 00000000 @04 -> + 00000000 @00
Debug: 5272 8823 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-762041
Debug: 5273 8828 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-767750
Debug: 5274 8828 riscv-013.c:302 scan(): 41b - 00000000 @04 -> + 00000000 @04
Debug: 5275 8828 riscv-013.c:599 execute_abstract_command(): command=0x231009
Debug: 5276 8828 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-767808
Debug: 5277 8834 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-773614
Debug: 5278 8834 riscv-013.c:302 scan(): 41b w 00231009 @17 -> + 00000000 @00
Debug: 5279 8834 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-773663
Debug: 5280 8840 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-779489
Debug: 5281 8840 riscv-013.c:302 scan(): 41b - 00000000 @17 -> + 00231008 @17
Debug: 5282 8840 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-779535
Debug: 5283 8846 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-785386
Debug: 5284 8846 riscv-013.c:302 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 5285 8846 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-785588
Debug: 5286 8852 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-791278
Debug: 5287 8852 riscv-013.c:302 scan(): 41b - 00000000 @16 -> + 10000001 @16
Debug: 5288 8852 riscv-013.c:313 scan():  ->  progbufsize=16 datacount=1
Debug: 5289 8852 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-791335
Debug: 5290 8858 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-797176
Debug: 5291 8858 riscv-013.c:302 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 5292 8858 core.c:931 jtag_execute_queue(): [TIME] enter: 1521723475-797362
Debug: 5293 8864 core.c:933 jtag_execute_queue(): [TIME] exit: 1521723475-803064
Debug: 5294 8864 riscv-013.c:302 scan(): 41b - 00000000 @16 -> + 10000001 @16
Debug: 5295 8864 riscv-013.c:313 scan():  ->  progbufsize=16 datacount=1

timsifive, 2018-03-22:
The 0.13 code hasn't really been optimized. The 0.11 code has, and also does a lot more caching of registers in general. Some day I mean to improve this, but there's always something else that needs doing.

What are you doing where single-word reads are a bottleneck?

Tong-Lin, 2018-03-23:
Actually, we read 32words in 15s; Is it normal for 0.13 version？Is there a faster method read from memory;

timsifive, 2018-03-23:
That is really slow (2.1 words/second).

I have a test that reads 15 individual words. Against spike it takes 836ms (18 words/second), and against an Arty board FPGA (using Olimex USB->JTAG adapter) image it takes 1070ms (14 words/second).

The software can be improved, and that will happen some day, but there may be some simple change you can make to get a faster speed sooner.

Are you reading consecutive words? If so, you can go much faster by telling OpenOCD to read a block of memory.

How are you connecting to your target?

Tong-Lin, 2018-03-24:
I reading from 0x30000 to 0x3007C, it's consecutive words; 
I start openocd with its config(annotated configs about flash in sifive-e31arty.cfg):
openocd -c "adapter_khz 10000" -f "interface/ftdi/olimex-arm-usb-tiny-h.cfg" -f "board/sifive-e31arty.cfg"

Can you paste the openocd d3 log while reading consecutive words?

Tong-Lin, 2018-03-24:
@timsifive 
I read from 0x30000 to 0x3007C, it's consecutive words;
I start openocd with its config(annotated configs about flash in sifive-e31arty.cfg):
openocd -c "adapter_khz 10000" -f "interface/ftdi/olimex-arm-usb-tiny-h.cfg" -f "board/sifive-e31arty.cfg"

Can you paste the openocd d3 log while reading consecutive words?

timsifive, 2018-03-26:
Here's a log of reading 0x400 bytes, consecutively. (This was done using the gdb dump command (`dump ihex memory /tmp/tmp1FrZQ5.ihex 0x80000000 0x80000400`), but there are other possibilities.)

```
Debug: 7025 2535 gdb_server.c:1416 gdb_read_memory_packet(): addr: 0x0000000080000000, len: 0x00000400
Debug: 7026 2535 target.c:2097 target_read_buffer(): reading buffer of 1024 byte at 0x80000000
Debug: 7027 2535 riscv.c:1671 riscv_set_current_hartid(): setting hartid to 0, was 1
Debug: 7028 2535 riscv-013.c:359 scan(): 41b r 00000000 @10 -> + 00000000 @00
Debug: 7029 2535 riscv-013.c:359 scan(): 41b - 00000000 @10 -> + 00010001 @10
Debug: 7030 2535 riscv-013.c:370 scan():  ->  hartsel=1 dmactive
Debug: 7031 2536 riscv-013.c:359 scan(): 41b w 00000001 @10 -> + 00000000 @00
Debug: 7032 2536 riscv-013.c:370 scan():  dmactive -> 
Debug: 7033 2536 riscv-013.c:359 scan(): 41b - 00000000 @10 -> + 00010001 @10
Debug: 7034 2536 riscv-013.c:370 scan():  ->  hartsel=1 dmactive
Debug: 7035 2536 riscv.c:1686 riscv_set_current_hartid(): Initializing registers: xlen=64
Debug: 7036 2536 riscv-013.c:1946 read_memory_progbuf(): reading 256 words of 4 bytes from 0x80000000
Debug: 7037 2536 riscv-013.c:670 execute_abstract_command(): command=0x321008
Debug: 7038 2536 riscv-013.c:359 scan(): 41b w 00321008 @17 -> + 00000000 @00
Debug: 7039 2536 riscv-013.c:359 scan(): 41b - 00000000 @17 -> + 00331009 @17
Debug: 7040 2537 riscv-013.c:359 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 7041 2537 riscv-013.c:359 scan(): 41b - 00000000 @16 -> + 0c000002 @16
Debug: 7042 2537 riscv-013.c:370 scan():  ->  progbufsize=12 datacount=2
Debug: 7043 2537 riscv-013.c:359 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 7044 2537 riscv-013.c:359 scan(): 41b - 00000000 @16 -> + 0c000002 @16
Debug: 7045 2537 riscv-013.c:370 scan():  ->  progbufsize=12 datacount=2
Debug: 7046 2537 riscv-013.c:359 scan(): 41b r 00000000 @05 -> + 00000000 @00
Debug: 7047 2538 riscv-013.c:359 scan(): 41b - 00000000 @05 -> + 00000000 @05
Debug: 7048 2538 riscv-013.c:359 scan(): 41b r 00000000 @04 -> + 00000000 @00
Debug: 7049 2538 riscv-013.c:359 scan(): 41b - 00000000 @04 -> + 00000000 @04
Debug: 7050 2538 riscv-013.c:1203 register_read_direct(): [0] reg[0x8] = 0x0
Debug: 7051 2538 riscv-013.c:670 execute_abstract_command(): command=0x321009
Debug: 7052 2538 riscv-013.c:359 scan(): 41b w 00321009 @17 -> + 00000000 @00
Debug: 7053 2538 riscv-013.c:359 scan(): 41b - 00000000 @17 -> + 00321008 @17
Debug: 7054 2539 riscv-013.c:359 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 7055 2539 riscv-013.c:359 scan(): 41b - 00000000 @16 -> + 0c000002 @16
Debug: 7056 2539 riscv-013.c:370 scan():  ->  progbufsize=12 datacount=2
Debug: 7057 2539 riscv-013.c:359 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 7058 2539 riscv-013.c:359 scan(): 41b - 00000000 @16 -> + 0c000002 @16
Debug: 7059 2539 riscv-013.c:370 scan():  ->  progbufsize=12 datacount=2
Debug: 7060 2539 riscv-013.c:359 scan(): 41b r 00000000 @05 -> + 00000000 @00
Debug: 7061 2539 riscv-013.c:359 scan(): 41b - 00000000 @05 -> + 00000000 @05
Debug: 7062 2540 riscv-013.c:359 scan(): 41b r 00000000 @04 -> + 00000000 @00
Debug: 7063 2540 riscv-013.c:359 scan(): 41b - 00000000 @04 -> + 00000000 @04
Debug: 7064 2540 riscv-013.c:1203 register_read_direct(): [0] reg[0x9] = 0x0
Debug: 7065 2540 program.c:33 riscv_program_write(): 0x7fffc3c47ec0: debug_buffer[00] = DASM(0x0000000f)
Debug: 7066 2540 riscv-013.c:359 scan(): 41b w 0000000f @20 -> + 00000000 @00
Debug: 7067 2540 riscv-013.c:359 scan(): 41b - 00000000 @20 -> + 00042483 @20
Debug: 7068 2540 program.c:33 riscv_program_write(): 0x7fffc3c47ec0: debug_buffer[01] = DASM(0x00100073)
Debug: 7069 2540 riscv-013.c:359 scan(): 41b w 00100073 @21 -> + 00000000 @00
Debug: 7070 2541 riscv-013.c:359 scan(): 41b - 00000000 @21 -> + 00440413 @21
Debug: 7071 2541 riscv-013.c:670 execute_abstract_command(): command=0x241000
Debug: 7072 2541 riscv-013.c:359 scan(): 41b w 00241000 @17 -> + 00000000 @00
Debug: 7073 2541 riscv-013.c:359 scan(): 41b - 00000000 @17 -> + 00321009 @17
Debug: 7074 2541 riscv-013.c:359 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 7075 2542 riscv-013.c:359 scan(): 41b - 00000000 @16 -> + 0c000002 @16
Debug: 7076 2542 riscv-013.c:370 scan():  ->  progbufsize=12 datacount=2
Debug: 7077 2542 riscv-013.c:359 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 7078 2542 riscv-013.c:359 scan(): 41b - 00000000 @16 -> + 0c000002 @16
Debug: 7079 2542 riscv-013.c:370 scan():  ->  progbufsize=12 datacount=2
Debug: 7080 2542 program.c:33 riscv_program_write(): 0x7fffc3c48030: debug_buffer[00] = DASM(0x00042483)
Debug: 7081 2543 riscv-013.c:359 scan(): 41b w 00042483 @20 -> + 00000000 @00
Debug: 7082 2543 riscv-013.c:359 scan(): 41b - 00000000 @20 -> + 0000000f @20
Debug: 7083 2543 program.c:33 riscv_program_write(): 0x7fffc3c48030: debug_buffer[01] = DASM(0x00440413)
Debug: 7084 2543 riscv-013.c:359 scan(): 41b w 00440413 @21 -> + 00000000 @00
Debug: 7085 2543 riscv-013.c:359 scan(): 41b - 00000000 @21 -> + 00100073 @21
Debug: 7086 2543 program.c:33 riscv_program_write(): 0x7fffc3c48030: debug_buffer[02] = DASM(0x00100073)
Debug: 7087 2544 riscv-013.c:359 scan(): 41b w 00100073 @22 -> + 00000000 @00
Debug: 7088 2544 riscv-013.c:359 scan(): 41b - 00000000 @22 -> + 00100073 @22
Debug: 7089 2544 riscv-013.c:1059 register_write_direct(): [0] reg[0x8] <- 0x80000000
Debug: 7090 2544 riscv-013.c:359 scan(): 41b w 00000000 @05 -> + 00000000 @00
[... cut ...]
Debug: 8599 2573 riscv-013.c:359 scan(): 41b - 00000000 @17 -> + 00331008 @17
Debug: 8600 2573 riscv-013.c:359 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 8601 2573 riscv-013.c:359 scan(): 41b - 00000000 @16 -> + 0c000002 @16
Debug: 8602 2573 riscv-013.c:370 scan():  ->  progbufsize=12 datacount=2
Debug: 8603 2573 riscv-013.c:359 scan(): 41b r 00000000 @16 -> + 00000000 @00
Debug: 8604 2574 riscv-013.c:359 scan(): 41b - 00000000 @16 -> + 0c000002 @16
Debug: 8605 2574 riscv-013.c:370 scan():  ->  progbufsize=12 datacount=2
```

Tong-Lin, 2018-03-27:
Ok, It's really faster than using cmd "x/32x 0x80000000"; Does not ocd use burst mode (or something else) while using cmd "x/32x" ?
And which cmd you use to load program to memory?

timsifive, 2018-03-27:
On Mon, Mar 26, 2018 at 10:18 PM, Tong Lin <notifications@github.com> wrote:

> Ok, It's really faster than using cmd "x/32x 0x80000000"; Does not ocd use
> burst mode (or something else) while using cmd "x/32x" ?
>
I don't know what x/32x uses. Your OpenOCD log will show you.

> And which cmd you use to load program to memory?
>
`load`

Tong-Lin, 2018-03-28:
> I have a test that reads 15 individual words. Against spike it takes 836ms

Which cmd you use to read individual words?

timsifive, 2018-03-28:
(gdb) p/x *((int*)0x10000000)
$1 = 0x94dd1db2
(gdb) p/x *((int*)0x1000004c)
$2 = 0xcf1e9271
(gdb) p/x *((int*)0x10000098)
$3 = 0x4a53fb4a
(gdb) p/x *((int*)0x100000e4)
$4 = 0x1c48bd50
(gdb) p/x *((int*)0x10000130)
$5 = 0xb0ce10ee
(gdb) p/x *((int*)0x1000017c)
$6 = 0x7598ad9d
(gdb) p/x *((int*)0x100001c8)
$7 = 0x76cf44e9
(gdb) p/x *((int*)0x10000214)
$8 = 0xed0c83f3
(gdb) p/x *((int*)0x10000260)
$9 = 0xf2cf3cf4
(gdb) p/x *((int*)0x100002ac)
$10 = 0xd1724b09
(gdb) p/x *((int*)0x100002f8)
$11 = 0x91ea9571
(gdb) p/x *((int*)0x10000344)
$12 = 0x68dd48ec
(gdb) p/x *((int*)0x10000390)
$13 = 0x89cdc0c3
(gdb) p/x *((int*)0x100003dc)
$14 = 0xcdcb0c8d
(gdb) p/x *((int*)0x100003fc)
$15 = 0x9643bdb9

​

Tong-Lin, 2018-03-30:
Thanks a lot;
And how do you get the execute time of `p/x` and `dump` cmd?

timsifive, 2018-03-30:
I read the openocd -d log after running those commands.
​

Tong-Lin, 2018-04-04:
Thanks for your help,
Read slow was caused by USB1.1->JTAG adapter;
I use a ubuntu virtual machine in VirtualBox without Extension Pack, which only support USB1.1

closed 2018-04-16 by Tong-Lin