riscv-collab/riscv-openocd #234: Unexpected "Remote connection closed" with GDB PIPE and RISC-V

ivankravets, 2018-04-03:
See https://github.com/gnu-mcu-eclipse/openocd/issues/25

/cc @palmer-dabbelt

timsifive, 2018-04-03:
Can you rerun this, adding `-d -l /tmp/openocd.log` to the OpenOCD command line, and then attach the full openocd.log file to this issue after reproducing the problem?

ivankravets, 2018-04-03:
[openocd.log](https://github.com/riscv/riscv-openocd/files/1873507/openocd.log)

timsifive, 2018-04-03:
It looks like gdb decided to disconnect after it received the list of registers from OpenOCD.

Where exactly did your gdb come from?

Also, in the original report you mentioned that it works on Linux. What platform are you using that it fails on?

ivankravets, 2018-04-03:
> Where exactly did your gdb come from?

* https://www.sifive.com/products/tools/
* https://bintray.com/platformio/dl-packages/toolchain-riscv/1.70200.0#files

> Also, in the original report you mentioned that it works on Linux

@palmer-dabbelt reported that it works on Linux yesterday.

> What platform are you using that it fails on?

macOS. See full steps:

```
$ ~/.platformio/packages/toolchain-riscv/bin/riscv64-unknown-elf-gdb
GNU gdb (GDB) 8.0.50.20170724-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin16.7.0 --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target remote | /Users/ikravets/.platformio/packages/tool-openocd/bin/openocd -d -l /tmp/openocd.log -c "gdb_port pipe" -f /Users/ikravets/.platformio/packages/tool-openocd/scripts/board/sifive-hifive1.cfg
Remote debugging using | /Users/ikravets/.platformio/packages/tool-openocd/bin/openocd -d -l /tmp/openocd.log -c "gdb_port pipe" -f /Users/ikravets/.platformio/packages/tool-openocd/scripts/board/sifive-hifive1.cfg
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-16:51)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 3 command.c:544 command_print(): debug_level: 3
Debug: 14 3 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_log_output /tmp/openocd.log
Debug: 15 3 command.c:143 script_debug(): command - log_output ocd_log_output /tmp/openocd.log
Remote connection closed
(gdb)
```

timsifive, 2018-04-04:
On Tue, Apr 3, 2018 at 3:03 PM, Ivan Kravets <notifications@github.com>
wrote:

> Where exactly did your gdb come from?
>
>
>    - https://www.sifive.com/products/tools/
>    - https://bintray.com/platformio/dl-packages/toolchain-riscv/
>    1.70200.0#files
>
> Which one did you use? Or are you saying the failure occurs with either
one?

ivankravets, 2018-04-04:
I already said, the problem is linked with Mac

timsifive, 2018-04-05:
Yes, but you linked to two sets of tools. Which of these two sets shows the problem?

Have you tried to start OpenOCD separately from gdb so it listens on a port instead of using the pipe mechanism?

Unfortunately I don't have easy access to a Mac to reproduce your problem.

ivankravets, 2018-04-05:
Both sets show the problem. 

Yes, it works separately without pipe. 

The main problem here if GDB process dies, openOCD will continue to work. In case with pipe, openOCD will die as only you kill GDB. that is the reason why “pipe” was added to OpenOCD. To make developers happy who use IDEs.

palmer-dabbelt, 2018-04-09:
@timsifive Do you have time to look at this?  It's blocking Ivan from getting Platform IO up and running.

timsifive, 2020-03-30:
Closing due to inactivity.

closed 2020-03-30 by timsifive