riscv-collab/riscv-openocd #242: Some OpenOCD Style check errors

mwachs5, 2018-04-18:
I get the following errors which are unrelated to my branch when running the style checker step:

```

mwachs5, 2018-04-18:
Again, I think the issue is that it doesn't make sense for users of this repo to apply their patches against anything but `riscv`, until we actually fix the `riscv` branch!

mwachs5, 2018-04-18:
So in my PR I changed the diff to just be against `origin/riscv` and it seemed to work fine. Not sure why you didn't do it that way?

timsifive, 2018-04-18:
The only actual style problem I see is in encoding.h. I've fixed those.

I assume the fespi.h and debug_defines.h problems come in because the checker script doesn't get enough context lines so doesn't see this is all in a comment itself.

mwachs5, 2018-04-18:
Ah, hm. The checking-the-patch seems kinda weird way to go about it then

timsifive, 2018-04-20:
On Wed, Apr 18, 2018 at 4:47 PM, Megan Wachs <notifications@github.com>
wrote:

> Ah, hm. The checking-the-patch seems kinda weird way to go about it then
>

I can see that, but that's how the OpenOCD check-in system does it. If we
increase the lines of context to a lot then for patches we'll end up
checking the entirety of every file that's touched in the patch, which
seems like decent behavior.

timsifive, 2018-04-20:
I'd like to merge this. If anybody runs into style issues that are due to lack of context in the diff, we can increase that number in .travis.yml.

PR #243 (2018-04-20): Clean up encoding.h, and add more context for style check
Fixes #242

Change-Id: I4d151350bf26bd3ea7733cb5247e4990fb487194
commit 5fa6dae9 (2018-04-20): Disable style check for now.

Fixes #242

Change-Id: I4d151350bf26bd3ea7733cb5247e4990fb487194
closed 2018-04-20 by timsifive