riscv-collab/riscv-openocd #253: riscv-011.c depends on idcode

TM1234, 2018-05-15:
I am surprised to see that some code in riscv-011.c depends on the idcode of the target which seems to make the implementation very SiFive specific rather that just depending on the DTM version. Why is this and does it mean that non SiFive RISC-V implementations based on this "old" implementation or on the equivalent underlying Rocket Chip code might need a code change in order to cater for their specific idcode(s)?

https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-011.c#L1499

mwachs5, 2018-05-15:
This seems like something that can/should easily be removed, by determining the correct idle value dynamically  (as v13 does).

timsifive, 2018-05-15:
Per the comment, this is to work around buggy cores that don't set the dtmcontrol idle field even though they should. Looking at the DTM version doesn't help you identify which cores need this work-around.

Is this causing you trouble?

mwachs5, 2018-05-15:
But idle is *always* just a hint/minimum. So it shouldn't really matter what it's set to.

TM1234, 2018-05-15:
I don't know for sure that this is causing us problems but we are having problems debugging with the latest build of openocd against a Microsemi RISC-V core that is based on a SiFive E31 (?) that has an idcode of 0x10e31913 and having noticed this core I wondered if it was relevant. I would have to investigate/debug  further to see whether or not this code is a problem in this context. 

But I have no idea if the specific check should apply to this core or not. In any case I'm very surprised that openocd code is dependent on a specific idcode as that does not seem the correct way to do things in general.

timsifive, 2018-10-26:
FWIW, this change was added in b04d5e882122bf4562af89f79e85253cc8fdab45, which is not very clear on why. I don't want to remove it for fear that it'll break something, and the code is completely harmless otherwise.

closed 2018-10-26 by timsifive