riscv-collab/riscv-openocd #271: src/helper/log.h: diffs from master

ilg-ul, 2018-06-19:
In the common `log.h` file I currently see the folowwing diffs from OpenOCD master:

```
diff --git a/src/helper/log.h b/src/helper/log.h
index 512bcc51..d60587f7 100644
--- a/src/helper/log.h
+++ b/src/helper/log.h
@@ -149,6 +149,8 @@ extern int debug_level;
  */
 #define ERROR_FAIL						(-4)
 #define ERROR_WAIT						(-5)
+/* ERROR_TIMEOUT is already taken by winerror.h. */
+#define ERROR_TIMEOUT_REACHED			(-6)
 
 
 #endif /* OPENOCD_HELPER_LOG_H */
```

I think that this change can be upstreamed.

ilg-ul, 2018-06-20:
I submitted the patch to Gerrit:

http://openocd.zylin.com/#/c/4567/

timsifive, 2018-06-20:
This feels like it really should be submitted along with the code that uses
it. But if it's accepted as-is then that's a nice bonus.

ilg-ul, 2018-06-20:
You can review it, and explain why it is needed, maybe this helps.

closed 2018-10-26 by timsifive