riscv-collab/riscv-openocd #278: OpenOCD assumes hartid and hart index are the same

timsifive, 2018-07-13:


mwachs5, 2018-07-13:
How does OpenOCD use hartid? I mean, how does this issue come up?

timsifive, 2018-07-13:
Currently upon startup OpenOCD will eg. display:
```
Info : Examined RISC-V core; found 2 harts
Info :  hart 0: XLEN=64, misa=0x8000000000141129
Info :  hart 1: XLEN=64, misa=0x8000000000141129
```

That's really the hart index, though. And the source uses hartid everywhere, and that should be clarified.

This is not high-priority, but I think it's worth tracking.

TommyMurphyTM1234, 2024-06-05:
> @en-sc is it still the case?

I don't think that this format of output occurs anymore.

> Currently upon startup OpenOCD will eg. display:
> 
> ```
> Info : Examined RISC-V core; found 2 harts
> Info :  hart 0: XLEN=64, misa=0x8000000000141129
> Info :  hart 1: XLEN=64, misa=0x8000000000141129
> ```

I can't see that the per-hart info is logged at all:
```
cd src/target/riscv

grep "LOG.*hart" *
riscv-013.c:			LOG_TARGET_ERROR(target, "Couldn't place hart in halt group 0. "
riscv-013.c:			LOG_TARGET_ERROR(target, "Couldn't place hart back in halt group %d. "
riscv-013.c:		LOG_TARGET_DEBUG(target, "Detected %d harts.", dm->hart_count);
riscv-013.c:		LOG_TARGET_ERROR(target, "No harts found!");
riscv-013.c:	LOG_TARGET_DEBUG(target, "Waiting for hart to come out of reset.");
riscv-013.c:		LOG_TARGET_ERROR(target, "No harts were prepped!");
riscv-013.c:	LOG_TARGET_DEBUG(target, "halting hart");
riscv.c:	LOG_TARGET_DEBUG(target, "prep hart, debug_reason=%d", target->debug_reason);
riscv.c:	LOG_TARGET_DEBUG(target, "halting all harts");
riscv.c:	LOG_TARGET_DEBUG(target, "Polling all harts.");
riscv.c:		LOG_TARGET_WARNING(target, "%d harts should remain halted, and %d should resume.",
riscv.c:	LOG_TARGET_DEBUG(target, "stepping hart");
riscv.c:		LOG_TARGET_ERROR(target, "Unable to step rtos hart.");
riscv.c:	LOG_TARGET_DEBUG(target, "Resuming hart, state=%d.", target->state);
riscv.c:		LOG_TARGET_ERROR(target, "Can't save register %s on a hart that is not halted.",

grep "LOG.*Exam" *
riscv-011.c:	LOG_INFO("Examined RISCV core; XLEN=%d, misa=0x%" PRIx64,
riscv-013.c:	LOG_TARGET_INFO(target, "Examined RISC-V core");
```

TommyMurphyTM1234, 2024-06-06:
> @aap-sc, I'm not sure what Tim meant by 'hart index' and 'hartid'.

I'm pretty sure that by "index" he meant the index of the array/list that OpenOCD uses (or maybe used in the past?) internally to store hart info whereas the "hartid" is the actual Hart identifier and the two may not be the same number. For example the spec says that on hart must have the id 0 but otherwise hart id's just have to be unique and are not necessarily contiguous or monotonically incrementing:

> Volume II: RISC-V Privileged Architectures V20211203 - 3.1.5 Hart ID Register mhartid
> 
> Hart IDs might not necessarily be numbered contiguously in a multiprocessor system, but at least one hart must
> have a hart ID of zero. Hart IDs must be unique within the execution environment.

Actually "at least" looks like an error in the spec since exactly one hart must have an id of zero.

tom-van, 2025-09-19:
Just a warning

> We have 4 similar concepts:
> 
>     1. Target index in target list (i.e. the one that can be used instead of target's name in `targets` TCL command).
>        
>        * specified by the order in which targets are created.
>        * starts from zero
>        * contiguous

If an OpenOCD config creates also other target types than riscv or uses more chips on one JTAG chain, then the last two statements may not be valid from the point of view of riscv target code.

>     * (b) One DM per TAP, multiple harts per DM.
>       
>       * `coreid`s should start from zero and be contiguous for each TAP.
>       * If a user mistakes and two harts on one DM get the same `coreid`... Well, let's not think about it.

It's quite easy to make this error (as coreid defaults to 0) and quite hard to find out what happened, because debugging almost works - just doesn't selects harts and uses usually hart0

> The cherry on top is, the whole `coreid` and `dbgbase` configuration is not necessary. Usecase where none of them is specified can be supported with just filling-up DMs on each TAP as requested (with target index (1) determining where each hart goes). IMHO, this is a nice and easy-to-use default.

Using target index (1) would go against OpenOCD principles. Imagine two chips on one board in JTAG chain.
Normally you would use board config like:
```
set CHIPNAME chip_one
source [find target/something.cfg]
set CHIPNAME chip_two
source [find target/some_riscv.cfg]
```
In such board the target some_riscv gets different defaults from target indices (1) than if used alone. Not a good idea.