riscv-collab/riscv-openocd #287: Print helpful error on permission denied

tmagik, 2018-08-14:
When OpenOCD can't open the /dev/bus/usb/whatever files because , it would be nice to print a link to the Udev config script to enable user access to the device

timsifive, 2018-08-15:
That would be cool, although making this so it's always relevant would be hard. udev itself only applies to Linux, while I bet its configuration might differ depending on the Linux distribution. The best I can think of right now is to the message depend on the device name that's being opened, and read /etc/os-release to check that the distro is one that's know to use udev scripts in a specific location.
It all feels a bit messy, though I agree that this would be a good thing to improve, since it will often bite new users.

timsifive, 2020-03-30:
Closing because I can't think of a good solution. This is also not riscv-specific.

closed 2020-03-30 by timsifive