riscv-collab/riscv-openocd #289: Question: status of upstreaming of RISC-V support?

TM1234, 2018-08-21:
Hope it's OK to ask this here rather than on the mailing list or elsewhere...

What is the status of the upstreaming of the RISC-V support to the master openocd project and the plan going forward?

As far as I can see from diffing the master openocd sources against the ones here there are still several discrepancies. E.g. the former does not include fespi.c or riscv_debug.c/h among other mods.

Is the upstreaming of RISC-V support an ongoing process?
Is there any ETA for when it will be finished or any planned milestones?
Will modifications/enhancements to the RISC-V support continue to be done on this github and then upstreamed for the forseeable future or will such mods be done on the master openocd project at some stage?
Any idea when?

Thanks a lot.

timsifive, 2018-08-22:
OpenOCD upstreaming is on-going. Most of the work has been upstreamed, with 2 notable exceptions:
1. `-rtos riscv` doesn't work with upstream.
2. The flash support we added has not been upstreamed yet.

I think it's like that we'll keep working on this branch when the process is done, but upstream that work regularly.

There is no timeline.

TM1234, 2018-08-22:
Thanks Tim - I appreciate the info,
So from time to time (a) changes here (riscv-openocd) get upstreamed to the master openocd github and (b) changes in the master openocd github are sync'd back to the riscv-openocd github?

timsifive, 2018-08-22:
Yes, although so far (a) has been very rare. There are 2 big changes left, and from then on they'll hopefully all be small, and we can make it happen more often.

TM1234, 2018-08-22:
OK - thanks a lot. 
I'll close this now since it was simply a question.

closed 2018-08-22 by TM1234