riscv-collab/riscv-openocd #299: Can not find input file: `Makefile.in'

xiusiwen, 2018-08-31:
When I installed the `riscv-tools` by running `./build.sh`, an error occurred:
 `Configuring project riscv-openocd`
`config.status: error: cannot find input file: 'Makefile.in'`

I really can not find the `Makefile.in` in `riscv-openocd`, and I do not know how to resolve it.
Please help.
Thanks in advance.

timsifive, 2018-08-31:
Usually this problem happens when your automake is older than 1.14. (See eg. https://github.com/riscv/riscv-tools/issues/209)

closed 2018-11-07 by timsifive

hyperion009, 2023-11-06:
@timsifive  after switching to automake 1.15,  the issue, "configure.ac:32: error: Macro PKG_PROG_PKG_CONFIG is not available. It is usually defined in file pkg.m4 provided by pakcage pkg-config", pop up. Did I miss something ?

TommyMurphyTM1234, 2023-11-06:
> @timsifive after switching to automake 1.15

Automake 1.15 is over 8 years old:

* https://savannah.gnu.org/news/?id=8137

Why not use something more up to date?

* https://savannah.gnu.org/news/?group_id=77

timsifive, 2023-11-07:
I don't know what the minimum working automake is. The README says 1.14. Did you clean out everything from the old automake/autoconf before doing your new build? Simplest is probably to do a fresh `git clone`.

hyperion009, 2023-11-08:
@timsifive @TommyMurphyTM1234  thank Tim & Tommy，I solved it after updating pkg-config to 0.29.2.