riscv-collab/riscv-openocd #303: in dmi_scan function 'in' and 'out' buff array size should be 9 bytes

gravikumar92, 2018-10-04:
in riscv-013.c dmi_scan function.
"in" and "out" buffer size are 8bytes (64bits). Buffer size should be 66bits.
it is corrupting scan_field structure.
Please change it to 9 byte array.

uint8_t in[8] = {0};
	uint8_t out[8];
	struct scan_field field = {
		.num_bits = info->abits + DTM_DMI_OP_LENGTH + DTM_DMI_DATA_LENGTH,
		.out_value = out,
		.in_value = in
	};

mwachs5, 2018-10-04:
What you're saying is true only if `abits = 32`. `abits` is often be much smaller than 32, but it _is_  represented by a 6-bit value, in which case even 9 bytes of buffer isn't enough. So this code should be corrected to make `in` and `out` big enough for the maximum possible value of abits which is `8(abits_max)+4(data)+1(op)` bytes

gravikumar92, 2018-10-05:
correct..
Max abits value is 64 bits. We need (64+32+2)bits/13-bytes of in & out.

timsifive, 2018-10-05:
That sounds right. @gravikumar92, are you able to put together a pull request for this change, including a code comment that explains how the max size is achieved?