riscv-collab/riscv-openocd #312: Is Pulpino implementation supported?

jaychokshi, 2018-10-19:
Does this openOcd risc-v SW port support Pulpino risc-v implementation?

Thanks,
Jay

timsifive, 2018-10-19:
AFAIK Pulpino uses a debug interface that's completely different from what will be the standard for RISC-V, and OpenOCD only supports the (future) standard.

jaychokshi, 2018-10-22:
Thanks for the response Tim. Appreciate it mate. Where I can find more info about this delta? Are there any link(s) to doc or msg. thread that I could refer to?

timsifive, 2018-10-22:
I'm not aware of anything that lists the differences. The spec that OpenOCD
implements is at
https://github.com/riscv/riscv-debug-spec/blob/master/riscv-debug-draft.pdf.
Pulpino did their work when the official effort was just getting off the
ground, and hasn't been part of it.

closed 2018-10-26 by timsifive