riscv-collab/riscv-openocd #316: dmstatus read before dmactivate is an issue in examine() function.

gravikumar92, 2018-10-25:
DMACTIVE activates the debug module. But examine() function reading the `dmstatus` before `dmactivate`.

	uint32_t dmstatus;
	if (dmstatus_read(target, &dmstatus, false) != ERROR_OK)
		return ERROR_FAIL;
	LOG_DEBUG("dmstatus:  0x%08x", dmstatus);
	if (get_field(dmstatus, DMI_DMSTATUS_VERSION) != 2) {
		LOG_ERROR("OpenOCD only supports Debug Module version 2, not %d "
				"(dmstatus=0x%x)", get_field(dmstatus, DMI_DMSTATUS_VERSION), dmstatus);
		return ERROR_FAIL;
	}

	/* Reset the Debug Module. */
	dm013_info_t *dm = get_dm(target);
	if (!dm->was_reset) {
		dmi_write(target, DMI_DMCONTROL, 0);
		dmi_write(target, DMI_DMCONTROL, DMI_DMCONTROL_DMACTIVE);
		dm->was_reset = true;
	}

timsifive, 2018-10-25:
I think this is correct behavior. The debug spec says that when dmactive is 0:
> The module’s state, including authentication mechanism, takes its reset values (the dmactive bit is the only bit which can be written to something other than its reset value).

So reading dmstatus when dmactive is 0 will get you the reset value for every field in dmstatus. The reset value for the version field still reflects the implemented version, so this should be safe.

Is this causing problems for you?

gravikumar92, 2018-10-26:
I confused with this statement : "_Implementations may use this bit to aid debugging, for example by preventing the Debug Module from being power gated while debugging is active._".

Can you explain little bit.

timsifive, 2018-10-26:
That's a good point. We should figure out what we want the specification to say first, and then I'll make OpenOCD do whatever we decide there. I've opened https://github.com/riscv/riscv-debug-spec/issues/395 to resolve the spec issue.

gravikumar92, 2018-10-31:
Ok..

en-sc, 2024-03-07:
The issue is not present anymore. AFAIU, it was fixed by 906635c4bda717884f87c0bd241f5c86a8238b41.

closed 2024-03-07 by en-sc