riscv-collab/riscv-openocd #332: JTAG does not use debugger's idle count

atthecodeface, 2018-11-18:
The debug spec provides for an idle count to be used after every dm access.

This is read into info->dtmcontrol_idle.

This is never used; it presumably should be a baseline for idle_count in riscv-013.c:dmi_scan

When I add in this value to idle_count, the openocd works correctly with hardware that sets this value.

timsifive, 2018-11-20:
You're right about the intent of dtmcontrol_idle. I just wrote a small patch to do what you suggest, but it breaks debugging of some SiFive cores on Arty. I want to understand that before merging anything.

mwachs5, 2018-11-20:
Didn't OpenOCD use to do this?

rbarzic, 2020-08-12:
Is this fixed ? I have the impression that the latest version still ignores dtmd_idle field...

timsifive, 2020-08-12:
This problem likely still exists.

cquartier, 2021-05-12:
can anyone elaborate on this problem for me? I'm trying to use the Arty A7 board's rs2232h for jtag so I don't need another external jtag tool in order to flash the freedom riscv core. I am getting:

`Error: Unsupported DTM version 9. (dtmcontrol=0x9)`

when trying to use openocd with the `sifive-e31arty-onboard-ftdi.cfg` board file. Is this a problem with openocd software, or a problem with the jtag hardware configuration in the freedom core?

timsifive, 2021-05-12:
That is totally unrelated to this ticket. It seems like the kind of problem you'd get if there are timing issues. First thing I'd try would be to reduce the adapter speed in your OpenOCD config.