riscv-collab/riscv-openocd #336: RVDF FPU F/D register described as D only

aka-sps, 2018-11-26:
https://github.com/riscv/riscv-openocd/blob/b911e07d75efbfe175657588a64e4ffa67d29ac3/src/target/riscv/riscv.c#L2498

The type of floating point register is set to double precision only, while the same register can be used for single precision values (with a NaN-boxing).
Please use `struct reg_data_type` with `.type_class = REG_TYPE_CLASS_UNION` and `struct reg_data_type_union` .

Thank you

timsifive, 2019-02-13:
How can I reproduce the problem, and confirm that it is fixed by your suggested change?

aka-sps, 2019-02-14:
You define the type of 64-bit register only as double precision https://github.com/riscv/riscv-openocd/blob/80ef54dba2411f9354b3793d5832c5d8ad871b4b/src/target/riscv/riscv.c#L2554, while this register can hold NaN-boxed single-precision value too.

Your implementation prints floating point registers only in hexadecimal.

The following definition allows you to output the value of the FPU register as a float / double (and hexadecimal too):

~~~~ c
static reg_data_type IEEE_single_precision_type = {.type = REG_TYPE_IEEE_SINGLE,.id = "ieee_single",};
static reg_data_type IEEE_double_precision_type = {.type = REG_TYPE_IEEE_DOUBLE,.id = "ieee_double",};
static reg_data_type_union_field FPU_S = {.name = "S",.type = &IEEE_single_precision_type};
static reg_data_type_union_field FPU_D = {.name = "D",.type = &IEEE_double_precision_type,.next = &FPU_S};
static reg_data_type_union FPU_D_or_S = {.fields = &FPU_D};
static reg_data_type FPU_D_reg_data_type = {.type = REG_TYPE_ARCH_DEFINED,.id = "Float_D_or_S",.type_class = REG_TYPE_CLASS_UNION,.reg_type_union = &FPU_D_or_S};
~~~~

Perhaps the NaN-box will have to be described as a separate field.

timsifive, 2019-02-19:
OK, but how can I test this? I assume there's some program or commands I can run through gdb to see the difference you describe.

aka-sps, 2019-02-19:
Now you can see in gdb:
~~~~
(gdb) i r $f0
Invalid register `f0'
(gdb) i r $fa0
fa0            0.073227885178676025     (raw 0x3fb2bf100bff666d)
(gdb) i r $f10
Invalid register `f10'
~~~~
You cannot see single precision values. The canonical names of registers are also absent.

Desired form:
~~~~
(gdb) i r $fa0
fa0            {D = 0x0123456789abcdef, S = 0x26a24800}        {D = 0.0089915120126600502, S = 1.35330714e+010}
(gdb) i r $f0
f0             {D = 0x0, S = 0x0}       {D = 0, S = 0}
~~~~

OpenOCD `reg f10 force` output should look similar.

PR #465 (2020-04-10): Expose FPRs as single and double for F and D.
If a hart support both F and D, then expose the FPRs as a union of float
and double. Fixes #336.

Change-Id: I3d4503bbf9281d6380c51259388cd01d399b94d6
commit 464407cf (2020-04-10): Expose FPRs as single and double for F and D. (#465)

If a hart support both F and D, then expose the FPRs as a union of float
and double. Fixes #336.

Change-Id: I3d4503bbf9281d6380c51259388cd01d399b94d6
closed 2020-04-10 by timsifive