riscv-collab/riscv-openocd #338: sprintf and strcpy are unsafe : buffer overflow is possible

aka-sps, 2018-11-30:
https://github.com/riscv/riscv-openocd/blob/42be17aed602405a8a36aa06671ea7566ec3f2e4/src/target/riscv/riscv-013.c#L350

Use `snprintf` (Note: on buffer overflow last symbol is not `'\0'`, need force)

https://github.com/riscv/riscv-openocd/blob/42be17aed602405a8a36aa06671ea7566ec3f2e4/src/target/riscv/riscv-013.c#L352

Use `strncpy` (Note: on buffer overflow last symbol is not `'\0'`, need force)

en-sc, 2024-01-22:
@aka-sps, I would like to close this. It is not an issue anymore. The code in question was changed by https://github.com/riscv/riscv-openocd/pull/892 and currently it is guaranteed that the buffer is sufficient.

en-sc, 2024-01-26:
@aka-sps, I'm closing this one. Feel free to reopen, if you have any concerns.

closed 2024-01-26 by en-sc