riscv-collab/riscv-openocd #357: /Host/Work/openocd/openocd.git/src/target/riscv/riscv-013.c:989: register_read_direct: Assertion `number != GDB_REGNO_S0' failed.

srey942, 2019-03-11:
Dear Experts,

I get the following error when I debug using openocd through eclipse : 

GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00404-g20463c28 (2018-01-23-12:24)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
adapter speed: 900 kHz
Info : J-Link V10 compiled Jul 28 2017 08:59:01
Info : Hardware version: 10.10
Info : VTarget = 1.819 V
Info : clock speed 900 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x100039d3 (mfg: 0x4e9 (<unknown>), part: 0x0003, ver: 0x1)
Info : dtmcontrol_idle=7, dmi_busy_delay=1, ac_busy_delay=0
Info : dtmcontrol_idle=7, dmi_busy_delay=2, ac_busy_delay=0
Info : [0] Found 1 triggers
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, 1 triggers
Info : Listening on port 3333 for gdb connections
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : JTAG tap: riscv.cpu tap/device found: 0x100039d3 (mfg: 0x4e9 (<unknown>), part: 0x0003, ver: 0x1)
openocd: /Host/Work/openocd/openocd.git/src/target/riscv/riscv-013.c:989: register_read_direct: Assertion `number != GDB_REGNO_S0' failed.

It would be really grateful if anyone would help me to solve or please let me know where I am going wrong.

The openocd version used here is 0.10.0-7.4 and the Eclipse Version is 2018-12 (4.10.0).

ilg-ul, 2019-03-11:
What device are you trying to debug? 

Your OpenOCD version is quite old, did you try a more recent one? 

https://github.com/gnu-mcu-eclipse/openocd/releases

timsifive, 2019-03-11:
This does sound like something I fixed in the past, but I don't remember the details. As Liviu suggests, can you try a recent build of OpenOCD? The main (and default) `riscv` branch of this repo should work well.

ilg-ul, 2019-03-11:
@srey942, also please note that the latest GNU MCU Eclipse OpenOCD uses only the upstreamed sources; I no longer manually apply patches from this fork, since they had negative impact on other targets; instead, I wait for Tim to upstream the changes, and I expect this to be a safer approach.

srey942, 2019-03-13:
Hi @ilg-ul @timsifive ,Thanks for your help. when I used the latest openocd version ,the error was fixed.

closed 2019-03-13 by srey942

ilg-ul, 2019-03-13:
> when I used the latest openocd version

did you use the latest GME binary (v0.10.0-11), or you built it from the sources available in this project?

srey942, 2019-03-13:
I used the latest GME binary(v0.10.0-11).